charlie0129/batt #1: Battery charge stuck at 100%

djkz, 2023-04-16:
On macbook air M2, when the daemon is running, the charge shows 100%, even when it's completely unplugged and been running off the battery for a while.

Any way to debug what's causing that, logs also show battery charge at 100%:
`time="2023-04-16 09:44:24.906" level=debug msg="batteryCharge=100, limit=75, chargingEnabled=true, isPluggedIn=false, maintainedChargingInProgress=false"`

charlie0129, 2023-04-16:
So you mean the daemon always get 100% charge but the actual battery charge is not 100%?

Since `batt` gets battery charge from a SMC key called `BUIC`, it could be M2 have a different key than M1. I think it can be solved by using another way to get battery charge.

PS, What version of `batt` are you using? Can you try the latest version?

djkz, 2023-04-16:
Using the latest version, after installing the daemon system constantly reports 100% charge,  on the Os X UI widget and using `pmset`.

I've ran it for a while on the battery just to make sure that it's not actually fully charged, that thing is a beast but pretty sure an hour of video would at least kill 1% of the charge.

charlie0129, 2023-04-17:
Hmm, that’s really strange. It shouldn’t be able to do that… 

So it only reports 100% charge AFTER installing the daemon? There are a couple of ways to debug came to my mind:

- Will uninstalling the daemon revert the behavior? If so, that’s something interesting.
- Have you tried rebooting, or resetting SMC, just to rule out some other issues.
- Can you run `batt status` to see the wattage of your battery to see how much it is actually discharging?

djkz, 2023-04-17:
Seems to work now, one thing that might have caused it is when I launched the daemon I was plugged in and at 100%, when starting on the battery it seems to pick it up correctly.

> Will uninstalling the daemon revert the behavior? If so, that’s something interesting.
it didn't immediately reset the percentage, but once it got charged up completely it seemed to start managing the power correctly again.

closed 2023-04-17 by djkz