charlie0129/batt #2: Better limitter

exidler, 2023-04-20:
Classical charge limiter have 2 levels:
1. level when stop charging (currently batt named limit)
2. level when start charging (less that 1.)

When we have battery charged up to limit 1.) and still does not discharged to level 2.) charging does not started
This logic prevent unnecessary small charing

Can you add this functionality?

charlie0129, 2023-04-20:
Actually, it do have this feature currently. It is just not configurable (can be added in the future). By when you set a limit to `x`, it will stop charging at `x+1` and start charging at `x-1`.

charlie0129, 2023-04-22:
Hi, configurable upper and lower limit is implemented in the lastest commits, see https://github.com/charlie0129/batt#upper-and-lower-charge-limit . The files below are the binary for the latest commits in case you want to test it now.

[batt-v0.1.0-beta.4-2-g7833398-checksums.txt](https://github.com/charlie0129/batt/files/11301068/batt-v0.1.0-beta.4-2-g7833398-checksums.txt)
[batt-v0.1.0-beta.4-2-g7833398-darwin-arm64.tar.gz](https://github.com/charlie0129/batt/files/11301071/batt-v0.1.0-beta.4-2-g7833398-darwin-arm64.tar.gz)

exidler, 2023-04-28:
As I can see working, but one bug:
If reboot (maybe restart daemon) in middle state (between min/max limit) - start charging, but ideally - not to start, because lower limit does not reached.

[batt.log](https://github.com/charlie0129/batt/files/11356029/batt.log)

charlie0129, 2023-04-29:
Yes, I see. 

The reason why your mac starts charging between min/max is that when your Mac restarts, the charging state is _always set to enabled_ by macOS (this behavior cannot be altered). So your mac is **always** charging after a reboot. When `batt` is started, it also sees charging being enabled. But at the same time, your battery charge is between min/max limit, `batt` will do nothing but keep the existing charging state (enabled) until the upper limit is reached, which is by design. So it will keep charging to the upper limit.

Restarting the daemon shouldn't have this problem because the charging state is kept even if daemon is down (until rebooting or other major power event happens).

There is one possible solution to solve this. We can store the charging state before rebooting and restore it after rebooting. Of course it may introduce more code complexity. But since it only affects rebooting (I think Macs don't reboot often) and the charging limit is still working (It will just charge to the upper limit, which is fine in my opinion, for a rare event.), I may not work on it in the near future. Of course, pull requests are welcomed.

exidler, 2023-04-29:
I agree with you.
Thank you, I think feature well implemented.

exidler, 2023-05-08:
I'm here again, charging with two thresholds working not as desired (imho):
1. battery discharged below lower limit
2. charger connected
3. charing started
4. going to sleep (charger still connected, battery level between low and high limit)
5. wake up from sleep
6. charging does not started 

here is part of log
```time="2023-05-08 04:25:53.824" level=debug msg="batteryCharge=74, lower=70, upper=85, chargingEnabled=true, isPluggedIn=false, maintainedChargingInProgress=false"
time="2023-05-08 04:26:13.826" level=debug msg="batteryCharge=75, lower=70, upper=85, chargingEnabled=true, isPluggedIn=false, maintainedChargingInProgress=false"
time="2023-05-08 04:26:33.830" level=debug msg="batteryCharge=75, lower=70, upper=85, chargingEnabled=true, isPluggedIn=false, maintainedChargingInProgress=false"
time="2023-05-08 04:26:53.832" level=debug msg="batteryCharge=75, lower=70, upper=85, chargingEnabled=true, isPluggedIn=false, maintainedChargingInProgress=false"
time="2023-05-08 04:27:13.835" level=debug msg="batteryCharge=76, lower=70, upper=85, chargingEnabled=true, isPluggedIn=false, maintainedChargingInProgress=false"
time="2023-05-08 04:27:33.837" level=debug msg="batteryCharge=76, lower=70, upper=85, chargingEnabled=true, isPluggedIn=false, maintainedChargingInProgress=false"
time="2023-05-08 04:27:53.841" level=debug msg="batteryCharge=76, lower=70, upper=85, chargingEnabled=true, isPluggedIn=false, maintainedChargingInProgress=false"
time="2023-05-08 04:28:13.843" level=debug msg="batteryCharge=77, lower=70, upper=85, chargingEnabled=true, isPluggedIn=false, maintainedChargingInProgress=false"
time="2023-05-08 04:28:33.845" level=debug msg="batteryCharge=77, lower=70, upper=85, chargingEnabled=true, isPluggedIn=false, maintainedChargingInProgress=false"
time="2023-05-08 04:28:53.847" level=debug msg="batteryCharge=77, lower=70, upper=85, chargingEnabled=true, isPluggedIn=false, maintainedChargingInProgress=false"
time="2023-05-08 04:29:01.217" level=info msg="system is going to sleep but charge limit is enabled, disabling charging just before sleep, and delaying next loop by 300 seconds"
time="2023-05-08 04:38:12.252" level=debug msg="system has finished waking up, delaying next loop by 120 seconds"
time="2023-05-08 04:43:05.457" level=debug msg="this maintain loop waited 287 seconds after being initiated, now ready to execute"
time="2023-05-08 04:43:05.458" level=debug msg="batteryCharge=77, lower=70, upper=85, chargingEnabled=false, isPluggedIn=false, maintainedChargingInProgress=false"
time="2023-05-08 04:43:25.460" level=debug msg="batteryCharge=77, lower=70, upper=85, chargingEnabled=false, isPluggedIn=false, maintainedChargingInProgress=false"
time="2023-05-08 04:43:45.463" level=debug msg="batteryCharge=77, lower=70, upper=85, chargingEnabled=false, isPluggedIn=false, maintainedChargingInProgress=false"
time="2023-05-08 04:44:05.467" level=debug msg="batteryCharge=77, lower=70, upper=85, chargingEnabled=false, isPluggedIn=false, maintainedChargingInProgress=false"
time="2023-05-08 04:44:25.468" level=debug msg="batteryCharge=77, lower=70, upper=85, chargingEnabled=false, isPluggedIn=false, maintainedChargingInProgress=false"
time="2023-05-08 04:44:45.471" level=debug msg="batteryCharge=77, lower=70, upper=85, chargingEnabled=false, isPluggedIn=false, maintainedChargingInProgress=false"
time="2023-05-08 04:45:05.473" level=debug msg="batteryCharge=77, lower=70, upper=85, chargingEnabled=false, isPluggedIn=false, maintainedChargingInProgress=false"
time="2023-05-08 04:45:25.478" level=debug msg="batteryCharge=77, lower=70, upper=85, chargingEnabled=false, isPluggedIn=false, maintainedChargingInProgress=false"```

charlie0129, 2023-05-08:
I think this is expected, unfortunately. Just keeping the lid open should solve this issue.

> when charging set a 'wakelock', so mac won't sleep when lid closed

There are two sleep modes: `idle` and `forced`. Closing the lid or manually choose the sleep option will cause a `forced` sleep, otherwise `idle`.

I have already delayed `idle` sleep when charging is enabled. So if you keep your lid open, it will work as expected.

But I can't do anything about `forced` sleep, just as the name suggests (without some serious unwanted hacking). macOS won't allow this. **Closing the lid will cause a `forced` sleep.** So this sleep cannot be prevented or delayed. The only thing I can do is to disable charging beforehand to prevent overcharging.

Another way to solve this is to disable the disable-charging-pre-sleep feature `batt disable-charging-pre-sleep disable`. However, by doing this, your mac may charge above the limit during sleep. This is probably not what you want.

exidler, 2023-05-09:
Idk, but according small research, there are working solution to prevent force sleep on lid close.
I have tested, and it works.

I will made impl for myself, but can make PR for control api.

charlie0129, 2023-05-10:
Interesting. Does preventing forced sleep come with other drawbacks? I know this can be done but I'm afraid of other side effects so I didn't dig deep.