charlie0129/batt #6: Unable to set higher level logs

wangzhenxing4, 2023-09-08:
First of all, I'm using automatic setup utility to install the batt (step 1), then go to "/Library/LaunchDaemons/cc.chlc.batt.plist", modify the file. Then use "sudo batt install" to install the daemon, I would like to ask what the normal procedure should be?

charlie0129, 2023-09-08:
The auto installation script should setup everything for you. You don’t need to modify the launch daemon. If you want to install it manually, you can refer to the manual installation instructions.

charlie0129, 2023-09-09:
Sorry I failed to notice the title. If you want to set higher level logs, the procedure would be:

1. unload daemon `sudo launchctl unload /Library/LaunchDaemons/cc.chlc.batt.plist`
1. edit `/Library/LaunchDaemons/cc.chlc.batt.plist` to set appropriate log level
2. load daemon `sudo launchctl load /Library/LaunchDaemons/cc.chlc.batt.plist`

charlie0129, 2023-09-27:
Hi, I added the steps to README. I am going to close this issue. If you need additional help, feel free to reopen.

closed 2023-09-27 by charlie0129