charlie0129/batt #7: Auto calibration support

uzza1hossain, 2023-09-30:
Thanks for the neat and clean package. 

It would be excellent to have a system where users can set an interval, let's say 7-days. Afterward, every 7 days, it will automatically charge up to 100 (no matter the limit define by user), then discharge to 15, and finally recharge to the user-defined limit.

The flow will be like.

1. Keep track of the last calibration.
2. Check if the last calibration was 7 days ago or more.
3. If it's been more than 7 days, set the limit to 100%.
4. When it reaches a 100% charge, disable the adapter.
5. When it drops to 15%, enable the adapter.
6. Set the limit to the user-defined value.
7. Update the last calibration time.

charlie0129, 2023-09-30:
Good suggestion. I think this feature can be split into two parts: 

1. a cronjob that does the calibration; 
2. do calibration (steps 3-6). for example, using command `batt calibrate`; 

Part 1 can be done using a external cronjob that calls `batt calibrate` every X days, or implement it in `batt`. But I would prefer to call `batt calibrate` manually instead of automatically to avoid auto-calibration when I really need to use my computer :) So I think we can implement part 2 first. 

To make it hassle-free, we need to consider edge cases like: what if the computer sleeps/restarts during calibration? will the intermediate state cause any problems? I haven't really planned yet, these are just some initial thoughts come to my mind.

uzza1hossain, 2023-10-01:
Yes. You can split this feature into two parts.
Implement the second part first and let's see how it's goes. There is no need of GUI for starting.
Just second part of feature, do calibration (steps 3-6) exposed via a command `batt calibrate`
Looking forward to next release 🥳

SkullEnemyX, 2025-04-05:
I am also in favour of this battery calibration feature as I noticed that continuously using Batt for few days does disturbs the battery calibration. Automating it would be beneficial.