charlie0129/batt #8: Unable to charge the iphone

reesecahn, 2023-10-10:
My Mac is connected to the display screen, but when the battery level is higher than the limit, my phone cannot charge while connected to the Mac.
Please tell me how to resolve this situation, it is very important to me, and I would greatly appreciate your help.

charlie0129, 2023-10-10:
batt shouldn't cause this issue. Can you please check is there any external issues?

Just to be sure. I tested your scenario a MacBook Air M1 (Monterey and Sonoma) and iPhone 15 Pro. I can't reproduce this issue.

reesecahn, 2023-10-10:
After I disabled disable-charging-pre-sleep, this issue disappeared. I speculate that this situation is caused by my Mac disabling charging before going to sleep.

charlie0129, 2023-10-10:
Your phone can't charge _during sleeping_ or _while awake_?

diyfixtool, 2023-10-27:
It may be a faulty charger, a problem with the charging cable, a problem with the mobile phone charging port, a problem with the mobile phone battery, or a problem with the system settings.

diyfixtool, 2023-10-31:
Check the charging cable and USB adapter for signs of damage, such as broken or bent prongs. If the internal battery of the mobile phone is broken, you need to replace the battery. The PHONEFIX official website provides battery repair and replacement services, so you can learn more if you need it.

closed 2024-09-18 by charlie0129