charlie0129/batt #9: Push to brew

MoienBowen, 2023-10-18:
Hi, it's a nice project with high user-friendly commandes.

I am wondering if you will push your tool to brew, then it will be much convenience for those who use brew as the package manager.

Thanks.

charlie0129, 2023-10-20:
Yes I agree with make brew packages.

But it requires some experience with making homebrew packages, which I don't have currently :) I will look into this when I am free. For now, I will mark this issue as `help wanted` so someone can pick this up (before I can).

Jerry1144, 2024-03-15:
Basically, Homebrew/Core wants to provide for command line tools that:

* Can be distributed by building on users' machines
* Has an Open license
* Acts in line with other unix tools (i.e. can be passed another prefix during install, and can tolerate finding its dependencies in $PATH and going through multiple symlinks, to name few)
* Is somewhat popular and actively maintained

A Formula can then be trivially written where I just fill in the source tarball, the obvious dependencies, and lines like `./configure`, `make`, `make install` as if I were building it myself. Homebrew's build system then handles things such as `codesign` (disable GateKeeper to use no more), `chmod` and packaging into Bottles for distribution.

So, yes, there needs to be instructions regarding how to *build* batt before it can be accepted in homebrew/core. Binary distributions would go to Homebrew/Casks but making an installer is even more tedious, and I don't think they accept anything that requires disabling Gatekeeper to work. 
***
I may otherwise help with writing the formula, if building batt is straightforward enough.

Actually, it seems *too straightforward*. As someone who doesn't know Golang at all, referencing [some other tool using Go](https://github.com/Homebrew/homebrew-core/blob/master/Formula/n/notify.rb) I'm surprised it appears all that is needed is a `go build ./cmd.go`??? I do not see where the prefix was passed (for [non-Go Formulae](https://github.com/Homebrew/homebrew-core/blob/f5320ce64fbdcc539a8d6371942cda4e6ce803a2/Formula/s/squid.rb#L44) there's a `./configure --prefix=$PREFIX`, which `notify` does not have), nor explicit directives to move the built binary into the prefix afterwards (usually handled by `make install`, while some Python tools [outline which file to copy](https://github.com/Homebrew/homebrew-core/blob/f5320ce64fbdcc539a8d6371942cda4e6ce803a2/Formula/n/ninja.rb#L39) from the Formula). Does `go build` look at $PREFIX on the fly?

There also appears to be multiple packages this project depends on. Does Go installation *come with* these packages specified in `go.mod`, or does Go download them during the build stage? Homebrew guidelines indicate those should be explicitly named in the Formula and downloaded by `brew`, so download stages and build stages can be separated (don't want download failures appear as build failures). Still, that was not done in many Go projects in homebrew/core, hmm…

charlie0129, 2024-03-18:
Thank you @Jerry1144 for making the formula!

> Does Go installation come with these packages specified in go.mod, or does Go download them during the build stage?

Go downloads them during the build stage. I am afraid you cannot separate them.

I have read the formula. It seems some changes need to be made in batt to you can avoid some hacks in the formula. For example, you need a separate control for `AllowNonRootAccess`. If I understood correctly, I can add these changes in the next release.

I don't know how the formula and the service thing work at all. I will try to look into it and give you some feedback.

Jerry1144, 2024-03-18:
> I don't know how the formula and the service thing work at all.

It's pretty simple: the service block tells `brew` this file (picked and modified from the build path) is a launchd.plist, and then `brew service [verb]` handles placing that file into /Library/LaunchDaemon/, calling `launchctl bootstrap` and the rest. Just existing infrastructure paralleling your installation routine…

except for one caveat: batt daemon holds its in-memory config as the ground truth, so changing `batt.json` manually does not work at all - that file merely gets overwritten by the daemon. If I let `brew services` handle the daemon registration, it'd be using default settings for `AllowNonRootAccess` once and for all.

> For example, you need a separate control for `AllowNonRootAccess`. If I understood correctly, I can add these changes in the next release.

I'm fine with most of settings requiring admin rights, yes, but to me `status` needing root access feels a bit counter-intuitive. Setting AllowNonRootAccess to false is only a temporary solution to that - a finer grained access control where `status` can work non-root but everything else cannot, is optimal. Admittedly, I have no idea how to do that. Feels like a complete rewriting of the IPC model.

Anyway, thank you for your incredible little tool. I'm only helping out, so don't feel pressured towards a next release.

charlie0129, 2024-03-19:
Thank you for your detailed explanation. It saves much time! I will look into it when I am free.

> a finer grained access control where status can work non-root but everything else cannot, is optimal

I agree with you :). It is better than the current solution, but yes, requires more work to do. So it may not be planned in the near future.

charlie0129, 2024-03-22:
I have read through your formula. It seems that it is working well.

>  batt daemon holds its in-memory config as the ground truth, so changing batt.json manually does not work at all - that file merely gets overwritten by the daemon.

The solutions is thath if you edit the config when batt daemon is stopped, it won't be overwritten and should work the next time the daemon is started.

Jerry1144, 2024-03-25:
With <KeepAlive> true, launchd automatically launches `batt daemon` immediately after it's killed; and as part of its shutdown process, batt overwrites the edited config with its own, right before the new itself reads that back.

I recall background daemons usually interpret SIGHUP as ["reload your config" signal](https://stackoverflow.com/q/4042201). Maybe you can implement that…

charlie0129, 2024-03-26:
> I recall background daemons usually interpret SIGHUP as "reload your config" signal. Maybe you can implement that…

Thank you for your suggestion! Sure.

Jerry1144, 2024-06-13:
It's merged. https://github.com/Homebrew/homebrew-core/pull/166340

charlie0129, 2024-09-18:
> I recall background daemons usually interpret SIGHUP as ["reload your config" signal](https://stackoverflow.com/q/4042201). Maybe you can implement that…

Finally I got some time for opensource... It's implemented in https://github.com/charlie0129/batt/commit/0fbc86f5a758e7251ea550ec70bf901d777d8cbb

majal, 2024-10-07:
I just tried `brew install batt` on Sequoia and it worked. Thanks for this work!

charlie0129, 2024-10-07:
v0.3.0 is released but homebrew is not updated yet because (sadly) I use nix on my working machine as my package manager instead of Homebrew, but I am working on it. I will release the new version on Homebrew when I found some Apple Silicon computer to install and test Homebrew on.

I will first release a new version v0.3.1 to address some pain in writing the formula by making batt more cofigurable. The formula should have less hacks. Then I will update the formula.

Maybe a more convenient solution is to let GitHub Actions automatically open PRs in homebrew-core repo once a new version is released. (leaving this for future work...)

charlie0129, 2024-10-08:
v0.3.1 is a release purely targeted to improve Homebrew integration. Now the Homebrew batt formulae can be made made less hacky https://github.com/Homebrew/homebrew-core/pull/193177

charlie0129, 2024-10-08:
The Homebrew formula should be fairly stable now and the installation instructions are mentioned as an official installation method.

Thank you to all the contributors, especially @Jerry1144 for bootstraping the formula.

closed 2024-10-08 by charlie0129