charlie0129/batt #10: Will it continue to charge when it is turned off, is this intentional

Chenyu23333333, 2023-11-26:
this is a good project,but i have some questions .The charge will continue when I turn it off, and when I turn it on again, the charge is over 80, which makes me have to disconnect the charger to make it lower than 80, although I can do this by the command sudo pmset -b sleep 0; sudo pmset -b disablesleep 1 to shut down the screen external display still works,(im using the external display ) is there a better solution?
thanks.

charlie0129, 2023-11-26:
If you mean you turned your computer off, then yes, charge will ALWAYS continue if your computer is powered off. This is by design (or enforced by Apple). My suggestion is that you put your computer to sleep instead of shutting down (Apple also don’t like power offs anyway).

BTW, “batt adapter disable” will unplug the power for you. If you don’t want to type the pmset commands, you can use some third party apps like Amphetamine which can prevent sleep with lid closed and power unplugged.

Chenyu23333333, 2023-11-26:
Thank you very much. That's very helpful.

closed 2025-03-20 by charlie0129