charlie0129/batt #11: Clamshell mode?

neumannjan, 2023-12-13:
Hi,

does this work with clamshell mode? I currently have charging at 100%, but I wish it to discharge to 80% despitebeing connected to the adapter. However, my laptop is in clamshell mode (closed and connected to external monitor), which, when I disable the adapter, it goes to sleep immediately.

charlie0129, 2023-12-13:
Yes, the charge limit should work as long as you plug the power supply in and adapter is not disabled. However, discharging is not possible (yet). This is due to Apple's limitations (not allowing clamshell mode with power unplugged). If you disable the adapter, it is the same as unplugging the power supply so Apple will not allow this. You can try some 3rd apps to overcome this limit (like Amphetamine) to allow clamshell mode with power unplugged. Maybe `batt` can integrate this in the future.

charlie0129, 2024-12-07:
To summary this issue (for anyone that happens to come by). 

Apple decides clamshell mode is only achievable if:

1. Power adapter is attached;
2. External display is on.

Failing to fulfill any of these will prevent clamshell mode from being enabled.

A common pitfall is when you are using clamshell mode and you unplugged the power adapter (either by pulling the power cord or `batt adapter disable`), making first requirement unmet, causing MacBook to go to sleep immediately.

Recently I looked into what other 3rd party App does to allow clamshell mode with power unplugged. Basically they just disable sleep entirely by `pmset -a sleep 0`, so your Mac never sleeps, even if it is in your bag, unplugged.

If we were to implement this feature, we must consider multiple edge cases. For example, you certainly don't want your Mac burning battery in your bag when travelling. Considering the risks, it is a low priority feature.