charlie0129/batt #13: Turn off Magsafe LED if limit is reached

Voyzxh, 2024-03-26:
I was wondering if we can choose between having the Magsafe LED in green or turning it off when limit is reached. It's a bit distracting to have it in green while battery is just maintained at 80%. Thank you.

charlie0129, 2024-03-26:
I think it is possible. I can add this control in the next release. 

In the mean time, you can try the attached file. It does not have a control. It just turns the LED off instead of turning green.

[batt-v0.2.1-dirty-darwin-arm64.tar.gz (batt with LED-off)](https://github.com/charlie0129/batt/files/14751160/batt-v0.2.1-dirty-darwin-arm64.tar.gz)

KenGoBlue, 2024-04-01:
I'd love to see an option to have the MagSafe LED turn off either when the limit is reached or when prevent-idle-sleep is active. Personally, I like seeing the LED turn green when it hits the limit, even if the limit is the 80% charge I've set — but having it turn off if charging is disabled when I close my MacBook lid would be a clearer indication of maybe this application's most misunderstood behavior.

charlie0129, 2024-04-02:
> but having it turn off if charging is disabled when I close my MacBook lid would be a clearer indication of maybe this application's most misunderstood behavior

Thank you. Great advice!

charlie0129, 2024-12-07:
This feature is implemented.

closed 2024-12-07 by charlie0129