charlie0129/batt #14: Continuing to charge in Sleep Mode

Fnr890, 2024-04-04:
Hi, 

This little script/app has been great so far - nice work on this! But I notice that my mac did sometimes continue to charge in sleep mode with lid off. I set mine between 60%, and one time it charged to 77% and another to 61%. I wonder what the underlying issue is. Thanks in advance!

charlie0129, 2024-04-05:
61% is normal (slight deviation is expected), but 77% is worth debugging . Can you attach the log when it reaches 77% if possible? (logs can be found in `/tmp/batt.log`

Fnr890, 2024-04-05:
here it is. When i noticed it went up to 77%, I unplugged the charger until it went down to 50%~.

 time="2024-04-01 06:58:02.177" level=debug msg="batteryCharge=58, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-04-01 10:18:58.177" level=debug msg="batteryCharge=58, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-04-01 11:34:09.277" level=debug msg="batteryCharge=58, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-04-01 11:40:02.771" level=debug msg="received kIOMessageSystemHasPoweredOn notification, system has finished waking up"
time="2024-04-01 11:40:02.771" level=debug msg="delaying next loop by 120 seconds"
time="2024-04-01 12:56:09.477" level=debug msg="batteryCharge=58, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-04-01 15:46:58.771" level=debug msg="batteryCharge=58, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-04-01 15:47:18.775" level=debug msg="batteryCharge=58, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-04-01 20:40:51.773" level=debug msg="batteryCharge=58, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-04-01 20:41:11.778" level=debug msg="batteryCharge=58, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-04-02 00:18:35.377" level=debug msg="batteryCharge=74, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-04-02 02:45:43.388" level=debug msg="batteryCharge=77, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-04-02 02:46:03.392" level=debug msg="batteryCharge=77, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-04-02 02:46:23.397" level=debug msg="batteryCharge=77, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-04-02 03:02:10.414" level=debug msg="batteryCharge=77, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-04-02 03:02:30.413" level=debug msg="batteryCharge=77, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-04-02 03:20:02.277" level=debug msg="batteryCharge=77, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-04-02 03:20:22.277" level=debug msg="batteryCharge=77, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-04-02 03:38:13.452" level=debug msg="batteryCharge=77, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-04-02 03:38:33.459" level=debug msg="batteryCharge=77, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-04-02 03:38:53.463" level=debug msg="batteryCharge=77, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-04-02 03:39:13.467" level=debug msg="batteryCharge=77, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-04-02 03:39:33.471" level=debug msg="batteryCharge=77, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-04-02 03:39:53.475" level=debug msg="batteryCharge=77, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-04-02 03:40:13.480" level=debug msg="batteryCharge=77, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-04-02 03:57:46.760" level=debug msg="batteryCharge=77, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-04-02 03:58:06.772" level=debug msg="batteryCharge=77, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-04-02 04:14:25.637" level=debug msg="batteryCharge=77, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-04-02 04:14:45.638" level=debug msg="batteryCharge=77, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-04-02 04:15:05.641" level=debug msg="batteryCharge=77, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-04-02 04:15:25.645" level=debug msg="batteryCharge=77, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-04-02 04:15:45.649" level=debug msg="batteryCharge=77, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-04-02 04:16:05.652" level=debug msg="batteryCharge=77, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
[... cut ...]
time="2024-04-04 17:52:59.177" level=debug msg="batteryCharge=61, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-04-04 19:16:02.277" level=debug msg="batteryCharge=61, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-04-04 19:24:22.377" level=debug msg="batteryCharge=61, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-04-04 19:41:02.577" level=debug msg="batteryCharge=61, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"

Jerry1144, 2024-04-21:
> time="2024-04-01 20:41:11.778" level=debug msg="batteryCharge=58, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
> time="2024-04-02 00:18:35.377" [...]

I guess system fell into a deeper sleep in between, and firmware/iBoot took over and decided it wants to charge. This would also explain why Mac always starts charging during power down and restarts, despite the OS (by `batt`) telling it to stop before logging out.

I'll see whether it holds off charging above 80% if my iPhone is powered down. Maybe even Apple themselves cannot make this to work.

charlie0129, 2024-04-21:
Sorry for the late reply, I was a bit busy lately.

I also noticed this strange behavior on my coworker's mac (on Sonoma), where the computer just stops batt (without notifying batt that it is going to sleep, which it should) and causing it to charge over the limit. I haven't figured out which sleep state it is, how it happened yet.

Thank you @Jerry1144 for your analysis. If the mac sleeps, batt should receive a message about the mac is going to sleep, which is missing from the logs. Maybe macOS is having a bug or it is not sleep at all. Will it be hibernation? I haven't tested it yet.

Jerry1144, 2024-05-16:
> I'll see whether it holds off charging above 80% if my iPhone is powered down. Maybe even Apple themselves cannot make this to work.

Sorry for the delay, but yes, even the "limit charge to 80" setting doesn't seem to work when powered down. So all the battery health functions are probably only maintained by the OS, and have no correspondence in the firmware at all, not even a flag for unconditional charge limits.

charlie0129, 2024-05-16:
That's sad that now Apple's implementation is OS-only too, unlike the old Intel MacBook that have this in the firmware.

charlie0129, 2024-10-08:
> Yes you are totally right. But one thing is missing, if the mac sleeps, batt should receive a message about the mac is going to sleep, which is missing from the logs. Maybe macOS is having a bug or it is not sleep at all.

I have set `ProcessType` to `Interactive` to see if it helps.

https://github.com/charlie0129/batt/blob/a7fedaf3ea1cceacf8b35c6d64277ee5a9d1963c/hack/cc.chlc.batt.plist#L9-L11

90michael, 2024-12-06:
Another log (v.0.3.3)
```
time="2024-12-05 11:57:13.468" level=debug msg="batteryCharge=70, lower=60, upper=70, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-12-05 11:57:33.471" level=debug msg="batteryCharge=70, lower=60, upper=70, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-12-05 11:57:53.475" level=debug msg="batteryCharge=70, lower=60, upper=70, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-12-05 11:58:13.478" level=debug msg="batteryCharge=70, lower=60, upper=70, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-12-05 11:58:33.483" level=debug msg="batteryCharge=70, lower=60, upper=70, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-12-05 11:58:50.115" level=debug msg="received kIOMessageCanSystemSleep notification, idle sleep is about to kick in"
time="2024-12-05 11:58:50.117" level=debug msg="batteryCharge=70, lower=60, upper=70, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-12-05 11:58:50.117" level=debug msg="no maintained charging is in progress, allow idle sleep"
time="2024-12-05 11:58:51.219" level=debug msg="received kIOMessageSystemWillSleep notification, system will go to sleep"
time="2024-12-05 11:58:51.220" level=info msg="charge limit is enabled, disabling charging, delaying next loop by 300 seconds, and allowing sleep"
time="2024-12-05 12:13:00.637" level=debug msg="received kIOMessageSystemHasPoweredOn notification, system has finished waking up"
time="2024-12-05 12:13:00.637" level=debug msg="delaying next loop by 120 seconds"
time="2024-12-05 12:14:44.914" level=debug msg="received kIOMessageCanSystemSleep notification, idle sleep is about to kick in"
time="2024-12-05 12:14:44.914" level=debug msg="system has just waked up (104.313969s ago), deny idle sleep"
time="2024-12-05 12:14:48.934" level=debug msg="received kIOMessageCanSystemSleep notification, idle sleep is about to kick in"
time="2024-12-05 12:14:48.936" level=debug msg="system has just waked up (108.335653s ago), deny idle sleep"
time="2024-12-05 12:15:03.222" level=debug msg="received kIOMessageCanSystemSleep notification, idle sleep is about to kick in"
time="2024-12-05 12:15:03.224" level=debug msg="system has just waked up (122.624248s ago), deny idle sleep"
time="2024-12-05 12:15:52.941" level=debug msg="received kIOMessageCanSystemSleep notification, idle sleep is about to kick in"
time="2024-12-05 12:15:52.941" level=debug msg="system has just waked up (172.341084s ago), deny idle sleep"
time="2024-12-05 12:16:50.539" level=debug msg="this maintain loop waited 298 seconds after being initiated, now ready to execute"
time="2024-12-05 12:16:50.541" level=debug msg="batteryCharge=70, lower=60, upper=70, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-12-05 12:16:56.963" level=debug msg="received kIOMessageCanSystemSleep notification, idle sleep is about to kick in"
time="2024-12-05 12:16:56.964" level=debug msg="system has just waked up (236.364952s ago), deny idle sleep"
time="2024-12-05 12:17:02.770" level=debug msg="received kIOMessageCanSystemSleep notification, idle sleep is about to kick in"
time="2024-12-05 12:17:02.770" level=debug msg="system has just waked up (242.170717s ago), deny idle sleep"
time="2024-12-05 12:17:10.544" level=debug msg="batteryCharge=70, lower=60, upper=70, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-12-05 12:17:30.547" level=debug msg="batteryCharge=70, lower=60, upper=70, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-12-05 12:17:50.552" level=debug msg="batteryCharge=70, lower=60, upper=70, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-12-05 12:18:00.929" level=debug msg="received kIOMessageCanSystemSleep notification, idle sleep is about to kick in"
time="2024-12-05 12:18:00.930" level=debug msg="batteryCharge=70, lower=60, upper=70, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-12-05 12:18:00.930" level=debug msg="no maintained charging is in progress, allow idle sleep"
time="2024-12-05 12:18:01.996" level=debug msg="received kIOMessageSystemWillSleep notification, system will go to sleep"
time="2024-12-05 12:18:01.996" level=info msg="charge limit is enabled, disabling charging, delaying next loop by 300 seconds, and allowing sleep"
time="2024-12-05 13:26:21.633" level=debug msg="this maintain loop waited 294 seconds after being initiated, now ready to execute"
time="2024-12-05 13:26:21.634" level=debug msg="batteryCharge=70, lower=60, upper=70, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-12-05 13:33:51.336" level=debug msg="received kIOMessageSystemHasPoweredOn notification, system has finished waking up"
time="2024-12-05 13:33:51.338" level=debug msg="delaying next loop by 120 seconds"
time="2024-12-05 13:35:51.592" level=debug msg="this maintain loop waited 117 seconds after being initiated, now ready to execute"
time="2024-12-05 13:35:51.594" level=debug msg="batteryCharge=70, lower=60, upper=70, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
[... cut ...]
time="2024-12-06 09:41:57.998" level=debug msg="batteryCharge=74, lower=60, upper=70, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-12-06 09:42:18.001" level=debug msg="batteryCharge=74, lower=60, upper=70, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-12-06 09:42:38.003" level=debug msg="batteryCharge=74, lower=60, upper=70, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-12-06 09:42:58.005" level=debug msg="batteryCharge=74, lower=60, upper=70, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
```

charlie0129, 2024-12-07:
@90michael Indeed, your log indicates batt is paused multiple times for no obvious reason, which is worth debugging. This is probably the OP is experiencing.

1. May I ask which macOS version are you using (by running `sw_vers`)?
2. Can you show the output of `sudo launchctl print system/cc.chlc.batt`?

But the battery charge is within the normal range (70->74 is fine). Battery charge tend to deviate a little due to calibration.

90michael, 2024-12-09:
sw_vers:
```
ProductName:		macOS
ProductVersion:		15.1.1
BuildVersion:		24B91
```
launchctl print:
```
system/cc.chlc.batt = {
	active count = 1
	path = /Library/LaunchDaemons/cc.chlc.batt.plist
	type = LaunchDaemon
	state = running

	program = /usr/local/bin/batt
	arguments = {
		/usr/local/bin/batt
		daemon
		--log-level=debug
	}

	stdout path = /tmp/batt.log
	stderr path = /tmp/batt.log
	default environment = {
		PATH => /usr/bin:/bin:/usr/sbin:/sbin
	}

	environment = {
		XPC_SERVICE_NAME => cc.chlc.batt
	}

	domain = system
	minimum runtime = 10
	exit timeout = 5
	runs = 1
	pid = 354
	immediate reason = speculative
	forks = 2
	execs = 1
	initialized = 1
	trampolined = 1
	started suspended = 0
	proxy started suspended = 0
	last exit code = (never exited)

	spawn type = interactive (4)
	jetsam priority = 40
	jetsam memory limit (active) = (unlimited)
	jetsam memory limit (inactive) = (unlimited)
	jetsamproperties category = daemon
	jetsam thread limit = 32
	cpumon = default
	probabilistic guard malloc policy = {
		activation rate = 1/1000
		sample rate = 1/0
	}

	properties = keepalive | runatload | inferred program | managed LWCR | has LWCR
}
```
I've noticed it 2 or 3 times now and I think it was always 74% (+ 4%). Not really a big issue. Thanks for looking into it!

charlie0129, 2025-02-06:
I added some additional checks in [v0.3.5-beta.1](https://github.com/charlie0129/batt/releases/tag/v0.3.5-beta.1) to mitigate this issue (macOS randomly stops batt or goes to sleep without notifying batt) ***to a certain level***. Since it's a macOS problem, I have no control over it. Hope the mitigation will work. v0.3.5-beta.1 is now a pre-release. Personally I will test v0.3.5-beta.1 for some time, if it works fine and no bug is reported, I will make it a release.

job-gordon, 2025-06-02:
I have the same issue.