charlie0129/batt #16: Sleepcallback.go: deregister for sleep/wake notifications during shutdown

Jerry1144, 2024-04-21:
By adding an additional info string:
```
diff --git a/sleepcallback.go b/sleepcallback.go
index a3bbcc8..6285739 100644
--- a/sleepcallback.go
+++ b/sleepcallback.go
@@ -161,5 +161,6 @@ func listenNotifications() error {
 	if int(C.ListenNotifications()) != 0 {
 		return fmt.Errorf("IORegisterForSystemPower failed")
 	}
+	logrus.Info("listenNotifications() has returned.")
 	return nil
 }
```
it is clear that `listenNotifications()` never returns in the happy path, not even during shutdown. This means the sleep listener CFRunLoop() was simply terminated, not given a chance to cleaned up: as per `pwr_mgt/IOPMLib.h`, the ports listening for sleep notifications "must" be closed by the caller. The code for cleaning up those also appeared in [QA1340](https://developer.apple.com/library/archive/qa/qa1340/_index.html) listing 4. It's a memory leak.

You may have to investigate passing around the CFRunLoopRef of that thread to the Go side, so as to CFRunLoopStop it during exit, after which `CFRunLoop()` returns and clean up codes can be run. Or attach another source that sets an exit flag. [Doesn't seem simple](https://forums.developer.apple.com/forums/thread/746610).

---

On a side note, IOPMLib.h mentions that kIOMessageSystemWillPowerOn is sent before most hardware has powered up, so if you print a log there, it would probably block until hard disks come back online. Maybe you can remove that callback.

charlie0129, 2024-04-22:
Thank you for your helpful suggestions. I will try to fix them.

yexua, 2024-04-25:
> 感谢您提供的有用建议。我会尽力修复它们。

看到了这个跟休眠有关系，我在使用batt这段时间，拔掉电源合盖休眠耗电好像有些高，不知道跟你们讨论的是否相关。
我使用该命令看到了一些日志
`pmset -g log | grep -e "Sleep.*due to" -e "Wake.*due to"
`
[image]

charlie0129, 2024-04-25:
@yexua I see. I think the log you provided is written by macOS and isn't related to `batt` (this project). The `Batt` appeared in the log is just a abbreviation of battery.

yexua, 2024-04-25:
> @yexua I see. I think the log you provided is written by macOS and isn't related to `batt` (this project). The `Batt` appeared in the log is just a abbreviation of battery.

我不是太懂这个，我的MBP是刚购买的，之前休眠待机耗电量几乎为1%。使用batt后，一天的休眠耗电量达到了4%左右，我不确认是否与这个有关系。

charlie0129, 2024-04-25:
> 我不是太懂这个，我的MBP是刚购买的，之前休眠待机耗电量几乎为1%。使用batt后，一天的休眠耗电量达到了4%左右，我不确认是否与这个有关系。

Is your 1% (100%->99%) or otherwise (e.g. 80%->79%)? If it 100%->99% then it is expected because that 1% about the same as 5% on other batter levels (Apple lies about battey percentage). batt shouldn't affect sleep/hibernation battery usage if you close the lid. Can you rule out other variables by uninstalling batt?

yexua, 2024-04-25:
> > 我不太懂这个，我的MBP是刚购买的，之前的休眠功耗几乎为1%。使用电池后，一天的休眠功耗达到了4%左右，我不确定与这个是否有关系。
> 
> 你的1%（100%->99%）还是其他（例如80%->79%）？如果它是 100%->99%，那么这是预期的，因为这 1% 与其他电池级别的 5% 大约相同（苹果在电池百分比方面撒了谎）。如果关闭盖子，batt 不应影响睡眠/休眠电池的使用。能否通过卸载batt来排除其他变量？

我可以试试卸载后观察一下

charlie0129, 2024-09-18:
This issue has been fixed in the following commit:

https://github.com/charlie0129/batt/commit/2d52d68d8416e50ab935ae43f31f8d5456bfc83c

Thanks @Jerry1144 for pointing out. Closing it.

closed 2024-09-18 by charlie0129