charlie0129/batt #17: Hope to add the general switch

LonePheasantWarrior, 2024-07-11:
Sometimes I need to go out and want to turn off the restricted charging function. Currently, I can only reset the charging upper limit to 100% and then adjust it to 80% after returning to the work post, which seems a bit cumbersome. It would be good if there was a switch to quickly turn off the batt function!

This software helps me a lot. Thank you

charlie0129, 2024-07-11:
I think this can be done by adding a subcommand, e.g. `batt disable`, which just sets the upper limit to 100% under the hood.

Of course, we may add a argument to `batt disable` like `--for=1d` to let it turn on again after 1 day to provide some niceties.

LonePheasantWarrior, 2024-07-11:
> I think this can be done by adding a subcommand, e.g. `batt disable`, which just sets the upper limit to 100% under the hood.
> 
> Of course, we may add an argument to `batt disable` like `--for=1d` to let it turn on again after 1 day to provide some niceties.

I noted that when I set the charging upper limit to 100%, the batt status shows `Charge limit: 100% (batt disabled)`, which is not bad either.

charlie0129, 2024-09-18:
> Of course, we may add an argument to `batt disable` like `--for=1d` to let it turn on again after 1 day to provide some niceties.

I think *sheduled re-enabling* like `--for=1d` is worth implementing, so I will reopen this issue.