charlie0129/batt #19: adapter disable stops external monitor

90michael, 2024-10-01:
Great project, thank you!
I connect my MacBook Pro (Apple M1 Pro) to an external monitor via Thunderbolt 3. The lid is closed and the only active screen is the external monitor. The external monitor is also charging the MacBook via the same Thunderbolt connection. As soon as I execute `sudo batt adapter disable`, the charging process stops as expected, but the external monitor no longer shows a picture. With an open lid the monitor shows a picture again.

charlie0129, 2024-10-01:
I assume this is  the same issue as https://github.com/charlie0129/batt/issues/11#issuecomment-1853925382

Unfortunately, this is a limitation of Apple's Clamshell mode. Although not recommended, you can use a third tool (mentioned in the above link) to override this limit with some caveats.

90michael, 2024-10-01:
Thanks for the quick reply. Yes, it seems to be the same issue. I have never heard the term Clamshell.