charlie0129/batt #20: sleep battery drain

jonas0b1011001, 2024-10-22:
Ever since installing batt my macbook looses about 2%/h while in sleep unplugged.

Did i misconfigure?

[image]

charlie0129, 2024-10-22:
Can you confirm that your MacBook is actually sleeping during that time?

You can check this by reading `/tmp/batt.log`. If adjacent logs are more than 20 seconds apart, then your mac is sleeping.

For example:

Sleeping:

```
time="2024-10-22 02:01:28.723" level=debug msg="batteryCharge=60, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-10-22 02:01:48.725" level=debug msg="batteryCharge=60, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-10-22 03:05:18.259" level=debug msg="batteryCharge=60, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-10-22 03:05:38.270" level=debug msg="batteryCharge=60, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-10-22 04:09:28.449" level=debug msg="batteryCharge=60, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-10-22 04:09:48.453" level=debug msg="batteryCharge=60, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-10-22 04:10:08.456" level=debug msg="batteryCharge=60, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
```

Awake:

```
time="2024-10-22 13:20:47.215" level=debug msg="batteryCharge=59, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-10-22 13:21:07.217" level=debug msg="batteryCharge=59, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-10-22 13:21:27.220" level=debug msg="batteryCharge=59, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-10-22 13:21:47.222" level=debug msg="batteryCharge=59, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
time="2024-10-22 13:22:07.226" level=debug msg="batteryCharge=59, lower=58, upper=60, chargingEnabled=false, isPluggedIn=true, maintainedChargingInProgress=false"
```

jonas0b1011001, 2024-10-22:
Logs are roughly 20s apart.

Can batt be preventing sleep or do i need to look somewhere else?

charlie0129, 2024-10-22:
batt will not preventing sleeping.

You can debug macOS sleep behavior by `pmset -g assertion`. Just be sure to open the same apps so you can see which one is preventing sleeping.

jonas0b1011001, 2024-10-22:
I will take a deeper look at that.

Thank you very much for the tool and your quick response!

closed 2024-10-22 by jonas0b1011001