charlie0129/batt #21: Question: does it automatically charges battery to 100% from time to time to keep it balanced?

tairosonloa, 2024-12-07:
Hey @charlie0129 !!

First, thank you (and all the contributors) for such a great piece of software. Great documentation too, I read it top to bottom. I love KISS and I had the same problem as you with other free or paid alternatives elsewhere.

After reading that great documentation, there is a piece of information I'm personally missing. Apple Optimized Battery Charging does, from time to time, charge the device up to 100% even when the limit is enabled. It does that, in Apple words (1), to ensure battery keeps calibrated and balanced.

Does this small utility does something similar, even with additional configuration? In case it doesn't, do you think it could be a great feature or something that is not really needed?

Thank you very much 🙇 

(1) I tried to find the official Apple article from support where I read it some time ago, but I couldn't find it.

charlie0129, 2024-12-07:
Thank you! Glad that you find batt to useful!

Regarding to recalibration, I think it is a good-to-have type of feature. It will serve a small portion of users.

- It definitely make the battery percentage more accurate if you are plugged in for an extended period of time and *not using the battery at all*. By saying "an extended period of time", I mean "over a month". I even encounterd it myself on my old Intel MacBook Pro, where battery will suddenly drop from 40% to 20% when I first use the battery after being plugged in for over 2 months straight as a desktop computer.
- However, for regular use cases, it is not a necessity. If you use your MacBook's battery from time to time (e.g. having meetings, on business trips), even if you only use 20% every several days, your battery shouldn't have a problem with calibration.

I think we can implement calibration in the future but not with high priority.

- Automatic calibration is handy but may surprise users when their MacBook suddenly uses battery power when plugged in.
- Manual calibration should be a better option for such a commandline utility in my opinion. It is predictable.

tairosonloa, 2024-12-07:
Thank you!

Yep, I'm the kind of person that uses the MBP as a desktop computer. I have it plugged all day and rarely disconnect its, like once very 1-2 months.

Okay, I may take a look to this in the future, and work on a PR. I'll inform you in this same issue if I plan to work on this seriously. Thank you again! 😄

charlie0129, 2024-12-07:
Oh, great! A PR is definitely appreciated.

BTW: I am planning a major refactor of the codebase in the near future to improve the modularity and testability of this old codease. I will try to finish this refactor faster if  you are to work on the code.

> Yep, I'm the kind of person that uses the MBP as a desktop computer. I have it plugged all day and rarely disconnect its, like once every 1-2 months.

Then calibration do have a strong reason for your desktop use case. (:

tairosonloa, 2024-12-07:
Don't rush it for me, mate. You know how contributing to FOSS is. I might do it next weekend, next month, next year, or forget it forever because maybe I don't find the time for it, and/or for me it's enough doing it manually (`batt adapter` or `batt disable` or whatever other command) every few weeks or once a month 😂 .

Maybe you don't hear from me in a while, and you decide to do it yourself, or maybe a third person comes here in the future and decides to do it.

So develop it at your own pace :) I'll inform you if I decide to work on it as I said, so you know it before I do anything in case you want to coordinate something or tell me something. 👍