charlie0129/batt #26: [Feature] Let it use battery

leornt, 2025-02-27:
Although the readme says 
`If your current charge is above the limit, your computer will just stop charging and use power from the wall. It will stay at your current charge level, which is by design.`

It's possible to add an option like `batt auto-drain enable` or something like that, that will automatically let the system drain battery until it reaches the lower delta or limit? This may be useful in cases where the Mac is used as desktop, and can also help to start the auto calibration feature (think the auto calibration is already mentioned in other issues).

Probably is a lot more complicated than just a `if (batteryLevel > limit) batt adapter disable` but... is it possible?

charlie0129, 2025-02-27:
Yes, it's possible, with extra considerations.

The reason why I didn't add this feature is that it has a problem when you use clamshell mode (which a lot of desktop users use): 

If adapter is disabled (to drain battery) in Clamshell mode, macOS will immediately go to sleep. The user have to either open the lid or plugin to use the computer again. This requirement will confuse users.

There is very hacky way to prevent sleeping in unplugged clamshell mode. You have to disable sleep globally by `pmset -a sleep 0`. I am hesitant to implement this hack because we need to consider every edge case to prevent potential problems. For example, when system is still draining and sleep is disabled, if the user decide to put the mac into his bag to travel, his mac will be running inside a bag, which is a bad idea.

If this feature were to be implemented, it will definitely need a lot of testing.

leornt, 2025-02-27:
Got it, as solving all the possible scenarios isn't a viable option right now, maybe letting the user choose between them is.

Presenting all those options can confuse the user, but adding "modes" gives it a simple use.

Want to use it as clamshell? `batt mode clamshell`. Want to use as a notebook but always plugged? `batt mode desktop`. Want to go back and use the intended defaults? `batt mode default`.

When choosing between those modes, present the user the drawbacks, like in clamshell mode, where they cannot discharge nor calibrate, and the default mode will keep the battery level ignoring the charger. Most of the users are fine with the defaults, and the ones that aren't will need to read the docs anyway.

Thank you for the quick feedback, and don't worry on rushing it, the app as it is, is already amazing and I'll keep using everyday

charlie0129, 2025-02-27:
Great suggestion about Modes to quickly switch among different configurations. Also allowing the user to add his own mode will add even more flexibility.

danbrice, 2025-04-09:
I'd like this feature too. I think a simple way to implement it would be to add a new `auto` option to the adapter setting. 

i.e. `batt adapter enable|disable|auto` 

On `auto` it will disable the adapter above the limit, and enable it at or below the limit. If there is no limit the adapter will be enabled. 

The documentation for the adapter setting already explains the impact of the adapter setting with clamshell mode.

The adapter setting should continue to default to `enabled` to avoid impacting those using clamshell mode.