charlie0129/batt #27: GUI or Menu Bar app like Al Dente

SkullEnemyX, 2025-04-05:
The reason why Al Dente is so famous is because of its ease of use. People would appreciate a barebones (if not a high quality) GUI or menu bar app to control everything by just a click. A more intuitive way would be to replace the native charging icon and have Batt's own charging icon so that this app could stay forever in the menubar. 
It should especially have:
~ A slider to set the battery charge level
~ Battery calibration button
~ Charge level, charge capacity and cycle count
~ Battery temperature monitoring
~ Automate button that irrespective of the battery percentage in the laptop, if the user has turned on the button, it won't charge the battery but run only on the charging cable (useful for laptops which are used as desktops most of the time, turning it off would charge it to either 100% or the battery level the user has set).

charlie0129, 2025-04-06:
Yes, a GUI is preferable, but as an infrastructure guy, I don't have the expertise or time to make a GUI. In addition to the features you mentioned, it must be a native app, i.e., not something like electron. Contributions are welcome.

If I have time in the future, I may use https://github.com/progrium/darwinkit to build a GUI. It's native and can be written in Go (so I don't have to learn ObjC or Swift just for this purpose). I have made some experiments with it last year and I believe is should work.

charlie0129, 2025-04-17:
See #28

SkullEnemyX, 2025-05-01:
https://github.com/lzt1008/powerflow

See this app for powerflow which graphically shows how the energy is moving into the battery or is consumed by the devices, looks very intuitive. If you do find someone, ask them to match this finesse, would look great.

SkullEnemyX, 2025-05-01:
Isn't it possible to leverage ai to create a GUI for this? China has so many amazing AI tools for coding like deepseek etc.

charlie0129, 2025-05-02:
> https://github.com/lzt1008/powerflow

The GUI looks good to me, but this one is a Web app. I don't think Electron/Tauri apps will match the goal of this project, which needs to be lightweight and native. It's great if anyone has written this as a separate frontend for batt, but I don't think it will be integrated with batt.

> Isn't it possible to leverage ai to create a GUI for this? China has so many amazing AI tools for coding like deepseek etc.

Yes, current AI can do simple things very well but not a native and well-designed GUI, which is too complex for AI. AI still has its way to go. In fact, the entire batt repo don't have a single line of AI-written code.

--- 

Regarding to GUIs, other opensource developers will be the main developer of it, if any. I will prioritize the control logic of batt (there are open feature requests that need to be worked on), after which, I will consider GUIs.

charlie0129, 2025-06-23:
Also in #46