charlie0129/batt #28: A prototype GUI for batt

clzoc, 2025-04-17:
https://github.com/clzoc/BattGUI

I've created a simple GUI prototype for batt that mimics Aldente Pro's interface. While the tool still needs refinement, I would like to receive suggestions and guidance during continued development, particularly regarding the known issue in README.md about not yet finding a suitable method to obtain valid real-time power adapter voltage measurements.

[image]

[image]

charlie0129, 2025-04-17:
Great work. How does it communicate with batt?

Regarding to power adapter voltage measurement, it's possible on Intel but I can't find a way to measure it on Apple Silicon too.

clzoc, 2025-04-17:
current communication way is naive，it start a sub process to execute limit set of batt and get the output to indicate the status

charlie0129, 2025-04-17:
I briefly read the code. It seems it runs batt cli to communicate. I think using Unix Domain Socket to communicate is a better way. The api design of it is a bit rough right now since it is originally design to be called by batt itself only. I can put in some work to make it easier for 3rd party programs to connect.

charlie0129, 2025-04-17:
Communicating using UDS looks like this. For example, using curl, to set charge limit

```
curl -Lv --unix-socket /var/run/batt.sock -XPUT http://localhost/limit --data "60"
```

clzoc, 2025-04-17:
absolutely, i agree with you. I will adapt the unix domain socket to improve the tool very soon

clzoc, 2025-04-17:
thank you, i will try it and report the feedback once I achieve the process

charlie0129, 2025-04-17:
The GUI looks great. We can collaborate a bit and make batt more user-friendly to those can't handle cli. But don't rush it, I will try to refactor the HTTP APIs in UDS to make it more coherent.

clzoc, 2025-04-17:
got it, I will contribute my code based on the status and plan of batt tool, thank you for your excellent work on batt

clzoc, 2025-04-19:
I have update the communicate method between GUI and batt, in ContentView.swift, line 93-96:
    pr.executableURL = URL(fileURLWithPath: "/bin/zsh")
    //pr.arguments = ["-c", "source ~/.zshrc; \(battURL.path)
    //pr.arguments = ["-c", "source ~/.zshrc; \(battURL.path) limit \(await Int(num))"]
    pr.arguments = ["-c", "source ~/.zshrc; curl -Lv --unix-socket /var/run/batt.sock -XPUT http://localhost/limit --data \(await Int(num))"]
         
Besides, I upload a DMG archive file which can be loaded and used by internet downloader. Currently, this DMG lacks a proper icon. I would like you to decide on the design and style of the icon.

charlie0129, 2025-04-19:
Looks like you are using a shell that calls curl which is forking subprocess like calling batt.

I was originally assuming something like https://github.com/jerrykrinock/UnixDomainSocketsDemo/blob/main/UnixDomainSockets/UDSClient.swift that uses the native swift code would be used to use UDS. I don't know anything about swift. But after some simple researching, it seems that there isn't any library that allows HTTP over UDS (besides hand-rolling yourself). If that's a problem, I can make a HTTP server that listens using a network socket (instead of Unix socket).

Regarding to the icon, since it's your app, you have full control over it. You can choose whatever fits. Icons have a connection to batteries seems good to me.

charlie0129, 2025-04-19:
but anyway, the UDS problem isn't a big deal.

clzoc, 2025-04-19:
Thank you for the detailed feedback and suggestions. I appreciate you pointing out the potential subprocess overhead from the current shell/curl implementation – I acknowledge that moving toward a native Swift approach for UDS communication would be more efficient. I'll thoroughly investigate the Swift-native methods (like the UDSClient example you referenced) and research HTTP-over-UDS possibilities before determining whether adapting the server to network sockets becomes necessary. Regarding the icon, your confirmation about the battery-themed direction is helpful. I'll continue refining the visual design accordingly and keep you updated on both fronts. 

   

 
I was originally assuming something like https://github.com/jerrykrinock/UnixDomainSocketsDemo/blob/main/UnixDomainSockets/UDSClient.swift that uses the native swift code would be used to talk to UDS. I don't know anything about swift. But after some simple researching, it seems that there isn't any library that allows HTTP over UDS (besides hand-rolling yourself). If that's a problem, I can make a HTTP server that listens using a network socket (instead of Unix socket).
 

 
 
I was originally assuming something like https://github.com/jerrykrinock/UnixDomainSocketsDemo/blob/main/UnixDomainSockets/UDSClient.swift that uses the native swift code would be used to talk to UDS. I don't know anything about swift. But after some simple researching, it seems that there isn't any library that allows HTTP over UDS (besides hand-rolling yourself). If that's a problem, I can make a HTTP server that listens using a network socket (instead of Unix socket).

charlie0129, 2025-04-19:
hi, I have added your GUI in the installation section in the README so others gets see your app https://github.com/charlie0129/batt/commit/390ad40181360b09165a10d2c1491cb811710c65

clzoc, 2025-04-22:
I added native code in the Objective-C file powerInfo.m to communicate with the Unix domain socket intermediate file /var/run/batt.sock. After some simple testing, the current issue is: if the user includes the allow-non-root flag during the installation of batt—for example, sudo batt install --allow-non-root-access—then the GUI can adjust the current charge limit using the slider. However, if the user installs batt with just sudo batt install, the GUI will encounter a "permission denied" error when attempting to adjust the charge limit.

For now, I haven’t come up with a good design to resolve this issue. Perhaps I can first document this scenario in README.md and update it later once a proper solution is found.

charlie0129, 2025-04-22:
A privileged helper can solve this. I am sure you have seen some apps that pop up alerts to install privileged helpers, this is the same thing.

charlie0129, 2025-04-22:
You can stick with forking right now, and save UDS for later optimizations. I will write a document with HTTP APIs in it so you know what APIs are available, but I am too busy doing my work to write it.

clzoc, 2025-04-22:
Got it! I'll give it a try on my own first, and we can discuss it when you're free.