charlie0129/batt #29: Homebrew service fails immediately

frothedoatmilk, 2025-04-23:
Hello! I'm having some issues with my Homebrew-based installation. I'm on a 2021 MPB, running MacOS 14.7.4. Here's what's happening:
I run `sudo brew services start batt`, the service reports started successfully, however it shuts down almost instantly after.

Contents of `/tmp/batt.log`:
```
... [a lot of copies of these two lines with timestamps from today, same commit hash and version]
time="2025-04-23T10:39:30-04:00" level=info msg="batt version v0.3.7 commit 0292e5913c404f5d752f72bd340e161e3030c28c"
time="2025-04-23T10:39:30-04:00" level=fatal msg="unexpected end of JSON input"
```

This started today after starting my computer. I believe it was working yesterday, although I'm not positive. It was definitely working last Friday (April 18th, 2025).

I've tried:
- `sudo brew services stop batt`
- `brew uninstall batt` and followed the `rm -rf` commands when the cellar failed to remove (it gets `chown`'d to root when `sudo brew services start batt` is run)

Let me know if there's any other diagnostic info I can provide!

charlie0129, 2025-04-23:
It seems the configuration file is broken.

Can you show the output of `cat /opt/homebrew/etc/batt.json`?

charlie0129, 2025-04-23:
Can you delete it by sudo rm and try again?

frothedoatmilk, 2025-04-24:
Upon starting my laptop this morning it seems like the issue had vanished before I removed this file...weird. Anyway, seems to be working! Are you OK with my raising a PR to make this error message a little clearer?

closed 2025-04-24 by frothedoatmilk

charlie0129, 2025-04-24:
PRs are welcome!

But I don't see how this error happens. That file won't magically got corrupted. Maybe outputting the file content in that fatal log can be helpful for debugging?

frothedoatmilk, 2025-04-24:
That sounds like a solid plan!

For me it happens if the file exists, but is empty. I'm not super experienced with Go or its JSON parser, but here's my repro steps. (I'm using the binary install now so I can test this, but this occurs on both installation types)

1. `sudo batt uninstall` if applicable
2. `sudo rm /etc/batt.json && sudo touch /etc/batt.json`
3. `sudo batt install` => JSON parse error

charlie0129, 2025-04-24:
The current logic handles nonexistent file (/etc/batt.json). If it doesn't exist, it will create a new one. I think for empty files, it can also delete it and create a new one.

https://github.com/charlie0129/batt/blob/8e5c1bb6ab3e4ac07dd8ca5d1d34fd8b17576706/conf.go#L51-L61

charlie0129, 2025-04-24:
If the files exists, to determine if it's empty, you need to read it, e.g. os.ReadFile().

Yes, empty string to unmarshal json will break the unmarshaller

frothedoatmilk, 2025-04-24:
Seems like we can just say if `b == []` (might not be valid syntax) we load and save the default config?

charlie0129, 2025-04-24:
You are right. I agree with you.

Regarding to the syntax, something like `len(bytes.TrimSpace(b)) == 0` will do.

frothedoatmilk, 2025-04-24:
#30