charlie0129/batt #34: MacOS26 Tahoe, `batt status` failed with code 500

CrackedPoly, 2025-06-10:
After update to MacOS 26 beta, `sudo batt status` fail to execute:

```
❯ sudo batt status                                                                                                                                              
Password:
Error: failed to get charging status: got 500: "key has no data, check if it is valid"
```

charlie0129, 2025-06-10:
Looks like the key for battery control is gone. I don't have immediate plans to upgrade to macOS 26 yet. Could you check the existing SMC keys on your mac using the following program? 

1. Download the attachment [here (smc.tgz)](https://github.com/user-attachments/files/20668721/smc.tgz), and extract it to `Downloads`
2. Open a terminal
3. `cd Downloads && ./smc -l | grep CH >smc-keys.txt 2>&1`
4. Open the newly created `smc-keys.txt` to see what's in it (or better, upload it to GitHub so we can debug it)

BTW: The program compiled from https://github.com/hholtmann/smcFanControl/tree/master/smc-command . If running a unknown program is a concern for you, you can compile it from the repo above.

CrackedPoly, 2025-06-10:
> Looks like the key for battery control is gone. I don't have immediate plans to upgrade to macOS 26 yet. Could you check the existing SMC keys on your mac using the following program?
> 
> 1. Download the attachment [here (smc.tgz)](https://github.com/user-attachments/files/20668721/smc.tgz), and extract it to `Downloads`
> 2. Open a terminal
> 3. `cd Downloads && ./smc -l | grep CH >smc-keys.txt 2>&1`
> 4. Open the newly created `smc-keys.txt` in `Downloads` to see what's in it (or better, upload it to GitHub so we can debug it)
> 
> BTW: The program compiled from https://github.com/hholtmann/smcFanControl/tree/master/smc-command . If running a unknown program is a concern for you, you can compile it from the repo above.

Here is the file: [smc-keys.txt](https://github.com/user-attachments/files/20670439/smc-keys.txt)

Or if you don't want to download it, actually it's small:
```
  B0CH  [si16]  0 (bytes 00 00)
  BCHT  [hex_]  (bytes 00)
  CH0D  [hex_]  (bytes 0e 00 00 00)
  CH0E  [hex_]  (bytes 00 00 00 00 00 00 00 00)
  CH0H  [ui32]  0 (bytes 00 00 00 00)
  CH0J  [ui8 ]  0 (bytes 00)
  CH0R  [hex_]  (bytes 00 00 00 00)
  CH0V  [ui32]  2805137408 (bytes a7 33 00 00)
  CHA1  [ui32]  1880162304 (bytes 70 11 00 00)
  CHA2  [ui32]  0 (bytes 00 00 00 00)
  CHAI  [ui32]  0 (bytes 00 00 00 00)
  CHAS  [ui32]  1880162304 (bytes 70 11 00 00)
  CHBI  [ui32]  0 (bytes 00 00 00 00)
  CHBV  [ui32]  1796276224 (bytes 6b 11 00 00)
  CHCC  [ui8 ]  1 (bytes 01)
  CHCE  [ui8 ]  1 (bytes 01)
  CHCF  [hex_]  (bytes 03)
  CHCR  [ui8 ]  1 (bytes 01)
  CHDB  [hex_]  (bytes 00 00)
  CHFS  [ui32]  16777216 (bytes 01 00 00 00)
  CHHC  [ui16]  0 (bytes 00 00)
  CHHV  [ui64]  (bytes 00 00 00 00 00 00 00 00)
  CHHW  [ui32]  0 (bytes 00 00 00 00)
  CHI1  [ui32]  1813053440 (bytes 6c 11 00 00)
  CHI2  [ui32]  4294967295 (bytes ff ff ff ff)
  CHIB  [ui8 ]  0 (bytes 00)
  CHIC  [ui8 ]  0 (bytes 00)
  CHIE  [hex_]  (bytes 00)
  CHIF  [ui32]  16777216 (bytes 01 00 00 00)
  CHIL  [flag]  (bytes 00)
  CHIM  [ui32]  2282946560 (bytes 88 13 00 00)
  CHIO  [flag]  (bytes 00)
  CHIS  [ui32]  16777216 (bytes 01 00 00 00)
  CHLS  [    ]  no data
  CHLT  [hex_]  (bytes 00 00 00)
  CHM2  [hex_]  (bytes ff ff ff ff 00 00 00 00 00 00 00 00)
  CHNC  [hex_]  (bytes 01 00 40 00 00 00 00 00)
  CHND  [hex_]  (bytes 00 00 00 00 00 00 00 00 00)
  CHNI  [si32]  (bytes 00 00 00 00)
  CHOC  [ui32]  0 (bytes 00 00 00 00)
  CHPS  [ui32]  16777216 (bytes 01 00 00 00)
  CHRT  [ui8 ]  0 (bytes 00)
  CHSC  [ui8 ]  0 (bytes 00)
  CHSE  [ui8 ]  0 (bytes 00)
  CHSL  [hex_]  (bytes 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00)
  CHST  [ui8 ]  0 (bytes 00)
  CHSW  [hex_]  (bytes 00 00 00 00 00 00 00 00)
  CHTC  [ui32]  0 (bytes 00 00 00 00)
  CHTE  [ui32]  0 (bytes 00 00 00 00)
  CHTL  [ui32]  269352960 (bytes 10 0e 00 00)
  CHTM  [ui32]  0 (bytes 00 00 00 00)
  CHTU  [ui32]  68157440 (bytes 04 10 00 00)
  TCHP  [flt ]  45 (bytes 00 80 32 42)
```

CrackedPoly, 2025-06-10:
> Thank your for your quick response.
> 
> Hmm, the key batt relies on to control charging is `CH0B` (or `CH0C`), which is missing from your report. It seems macOS 26 decides to remove this functionality for whatever reason.
> 
> Now the situation becomes a bit tricky because we need to find a new way to control the battery, but Apple never tells developers how to do it. Developers need to do some reverse engineering to find it out.

Only Apple can do.🤓

kahakaha, 2025-06-10:
> To anyone that came across this issue:
> 
> Warning
> 
> macOS Tahoe 26 is not supported because Apple has removed the functionality that batt relies on to control the battery. Until a new battery controlling method is found, we don't recommend upgrading to macOS Tahoe.
> 
> You can read the comment above for details: [#34 (comment)](https://github.com/charlie0129/batt/issues/34#issuecomment-2958535441)

Do you think it is a good idea to put this alert into README, too? I don't think people will click issues before they download `batt`.

mhaeuser, 2025-06-14:
The API is guarded behind the ``com.apple.private.iokit.soc-limit`` entitlement... Sigh.

mhaeuser, 2025-06-16:
@charlie0129 I saw that SMC key during RE, but in Sequoia even. I remember someone mentioning something alike was added, maybe this was some transition period to make sure old OSes can still handle new firmware (somewhat).

How does `com.apple.private.applesmc.user-access` work? Currently, we can just connect to SMC from userland and r/w (I think you do the same with `CH0C`, but I’m on phone, so can’t check) - are only some keys guarded or is the entire access now protected?

Thanks!

mhaeuser, 2025-06-16:
@charlie0129 Good - so I guess our tools are dead. Thanks! :))

Off-topic, I wonder why keys are individually protected. What good reason is there for a casual userland app to interface with SMC? Maybe also just a transition period, or grace for stuff like fan monitoring/control.

charlie0129, 2025-06-16:
> why should a random (well, even if root, really) app be allowed to interface with core hardware?

really? I thought you need to be root to write to the SMC. or did I miss something?

charlie0129, 2025-06-16:
> only drivers should interface with core hardware

sure, making sure only the kernel mode code can access it makes sense from a security standpoint

maybe just my experience with Linux affected my preference

mhaeuser, 2025-06-16:
@charlie0129 If you want to check further - there is a manual charge limit mode that's not really wired up it feels like (MCL) and a desktop mode with override to prevent charging beyond 80 % once on hold (I think) in PowerUI, and there are XPC shenanigans.

mhaeuser, 2025-06-17:
Sure. Honestly, I already dumped more time than I can afford right now and hoped someone else would do the work. :D

This stack is a lot more complicated than I thought (and than people make it sound like). There are various triggers for a full charge (periodic, DOD0, QMax), it drains in all modes (never heard this was possible), there’s desktop mode (??), several kinds of predictors, clean energy charging, and the manual mode.

Currently, manual mode, in one place, is used in combination with checks for the features “augury” and “hyperion”. “augury” is marked as “FeatureComplete” in Sequoia and I hope this is the slider we were thinking about. There’s a realistic chance we will get something like that later in the Tahoe beta cycle, but who knows.

If we won’t get it, later this summer will be much more pleasant to play with the XPC API. I hope I won’t get tempted before. :)

zipos, 2025-06-17:
> > there is a manual charge limit mode that's not really wired up it feels like (MCL) and a desktop mode with override to prevent charging beyond 80 % once on hold (I think) in PowerUI, and there are XPC shenanigans.
> 
> well it seems a bit hard, especially since i am not really good at reverse engineering :). till this point, i would prefer just wait and see. i won't upgrade to tahoe in the next 1 or 2 years so i am not in a hurry :P

Why not upgrade to stable release? Also AlDente Beta that works with Tahoe was released and you could try to reverse engineer that ;)

mhaeuser, 2025-06-17:
> Also AlDente Beta that works with Tahoe was released and you could try to reverse engineer that ;)

@zipos Thanks for the note! I might look at their fix soon. But AlDente is commercial and if they figured out something smart, let them have it to boost their product - I feel like “leaking” their idea would be unfair now, unless it’s something obvious after all.

mhaeuser, 2025-06-19:
@gh10ltz Uhhh… No, I understand things perfectly fine. Also, @charlie0129 already said he was not interested in approaching this now. You can give it a shot if you want to…

Considering Apple uses entitlements throughout the stack now, I’m not sure AlDente’s fix will be long-lived anyway.

Teddy190827, 2025-06-21:
why I can still use all batt commands in terminal without receiving the message Error: failed to get charging status: got 500: "key has no data, check if it is valid"
and while batt commands are available, it's not actually functioning, Mac just charges to full

[image]

charlie0129, 2025-06-21:
> why I can still use all batt commands in terminal without receiving the message Error: failed to get charging status: got 500: "key has no data, check if it is valid"

Try viewing `/tmp/batt.log` for errors. Also, it looks like you are using a very old version of batt.

oliver408i, 2025-06-24:
Im on the latest Tahoe beta, and with SIP and AMFI disabled, the common `smc` binary with `com.apple.private.applesmc.user-access` is able to access the `CHLS` key (not `CH0B` or `CH0C`, maybe thats a different entitlement). As far as I can tell `CHLS` is able to work. 
FYI for those who want to try this, the command is `sudo ./smc -k CHLS -w XX01` where XX is the hex of the charge limit. For 80 that is 50, so `5001`
Would still be interesting to try to figure out what aldente is doing, but it seems like for now this is as far as it will get. I don't know if Aldente will continue to work after a few updates but we'll see. I did a little string sniffing in the aldente binary and helper, and I can't find any new or different SMC keys it seems to be using, so whatever it's doing is probably a big workaround that is beyond most of us here.

charlie0129, 2025-06-25:
@oliver408i Yes. Both the `CHLS` and what mhaeuser's IOKit findings will work. Regarding to `CH0B/C`, I am pretty sure they are gone, not permission issues. Not sure what AlDenete is doing to make it work.

pickedbrian, 2025-07-02:
May I ask when batt will support macOS 26 Tahoe? Is there a development plan? BatteryOptimizer_for_Mac already supports it. I am considering switching to it, but I still prefer batt.  [https://github.com/js4jiang5/BatteryOptimizer_for_Mac/issues/44](url)

oliver408i, 2025-07-03:
> May I ask when batt will support macOS 26 Tahoe? Is there a development plan? BatteryOptimizer_for_Mac already supports it. I am considering switching to it, but I still prefer batt. [https://github.com/js4jiang5/BatteryOptimizer_for_Mac/issues/44](url) [@charlie0129](https://github.com/charlie0129)

Yep I saw that. Im not a big fan of that program due to the amount of additional stuff it adds, so I took the keys and made my own short script.
From that repo, I can see `CH0J` is the new discharging key (0 = off, 1 = on).`CHTE` controls charging. `00000000` is on/default behavior. `01000000` seems to be override disable. Both of these work in my testing.

charlie0129, 2025-07-03:
@pickedbrian @oliver408i Oh, new keys has already been discovered? That will significantly speed up development, making Tahoe support trivial.

I will add support to Tahoe if I have time. PRs are welcome. I cannot guarantee the exact time when tahoe support will come but it should before the official launch of tahoe.

The only roadblock is that I will not personally upgrade my macs to Tahoe, and we will rely on community to test if it will work.

oliver408i, 2025-07-03:
```
#!/bin/zsh

# --------- Detect if Apple Silicon ---------
get_cpu_type() {
  if system_profiler SPDisplaysDataType | grep -qi "chip.*Apple"; then
    echo "apple"
  else
    echo "intel"
  fi
}

# Abort if not Apple Silicon
if [[ $(get_cpu_type) != "apple" ]]; then
  echo "❌ This script is intended for Apple Silicon Macs only."
  exit 1
fi

# --------- Key Detection ---------
has_key() {
  smc -k "$1" -r 2>&1 | grep -qv "no data"
}

get_smc_numeric_value() {
  key="$1"
  smc -k "$key" -r 2>/dev/null | awk '{for (i=1; i<=NF; i++) if ($i ~ /^[0-9]+$/) { print $i; exit }}'
}

# Charging keys
has_CH0B=$(has_key CH0B && echo true || echo false)
has_CH0C=$(has_key CH0C && echo true || echo false)
has_CHTE=$(has_key CHTE && echo true || echo false)

# Discharging keys
has_CH0I=$(has_key CH0I && echo true || echo false)
has_CH0J=$(has_key CH0J && echo true || echo false)
has_ACLC=$(has_key ACLC && echo true || echo false)

# --------- Charging Control ---------
enable_charging() {
  echo "🔌 Enabling charging"
  if $has_CH0B; then sudo smc -k CH0B -w 00; fi
  if $has_CH0C; then sudo smc -k CH0C -w 00; fi
  if ! $has_CH0B && $has_CHTE; then sudo smc -k CHTE -w 00000000; fi
}

disable_charging() {
  echo "⛔ Disabling charging"
  if $has_CH0B; then sudo smc -k CH0B -w 02; fi
  if $has_CH0C; then sudo smc -k CH0C -w 02; fi
  if ! $has_CH0B && $has_CHTE; then sudo smc -k CHTE -w 01000000; fi
}

# --------- Discharging Control ---------
enable_discharging() {
  disable_charging
  echo "🔋 Forcing discharging"
  if $has_CH0I; then sudo smc -k CH0I -w 01; fi
  if $has_CH0J && ! $has_CH0I; then sudo smc -k CH0J -w 01; fi
  if $has_ACLC; then sudo smc -k ACLC -w 01; fi
}

disable_discharging() {
  echo "🔌 Stopping forced discharging"
  if $has_CH0I; then sudo smc -k CH0I -w 00; fi
  if $has_CH0J && ! $has_CH0I; then sudo smc -k CH0J -w 00; fi
  if $has_ACLC; then sudo smc -k ACLC -w 00; fi
}

# --------- SMC Status ---------
get_smc_charging_status() {
  if $has_CH0C; then
    val=$(get_smc_numeric_value CH0C)
    [[ "$val" == "00" ]] && echo "enabled" || echo "disabled"
  elif $has_CHTE; then
    val=$(get_smc_numeric_value CHTE)
    [[ "$val" == "0" ]] && echo "enabled" || echo "disabled"
  else
    echo "unknown"
  fi
}

get_battery_info() {
  pmset -g batt | grep -Eo "\d+%" | head -1
}

get_power_source() {
  pmset -g ps | head -n 1 | sed -E "s/Now drawing from '([^']+)'/\1/"
}

is_discharging() {
  if $has_CH0I; then
    val=$(get_smc_numeric_value CH0I)
    [[ "$val" == "01" ]] && return 0
  elif $has_CH0J; then
    val=$(get_smc_numeric_value CH0J)
    if [[ "$val" =~ ^[0-9]+$ ]] && [ "$val" -ne 0 ]; then return 0; fi
  elif $has_ACLC; then
    val=$(get_smc_numeric_value ACLC)
    [[ "$val" == "01" ]] && return 0
  fi
  return 1
}

# --------- CLI Interface ---------
case "$1" in
  charge|+c)
    if is_discharging; then disable_discharging; fi
    enable_charging
    if [[ $(get_power_source) != "AC Power" ]]; then
      echo "⚠️  Warning: No power adapter detected. The battery won't actually charge"
    fi
    ;;
  stop-charge|-c)
    disable_charging
    ;;
  discharge|+d)
    enable_discharging
    ;;
  discharge-off|-d)
    disable_discharging
    ;;
  status|s)
    echo "Battery Level:       $(get_battery_info)"
    power_source=$(get_power_source)
    case "$power_source" in
      "AC Power") power_icon="🔌";;
      "Battery Power") power_icon="🔋";;
      *) power_icon="❓";;
    esac
    echo "Power Source:        $power_icon $power_source"
    charge_status=$(get_smc_charging_status)
    [[ "$charge_status" == "enabled" ]] && charge_icon="✅" || charge_icon="❌"
    echo "SMC Charging:        $charge_icon $charge_status"
    if is_discharging; then
      discharge_icon="✅"
      discharge_status="yes"
    else
      discharge_icon="❌"
      discharge_status="no"
    fi
    echo "SMC Discharging:     $discharge_icon $discharge_status"
    ;;
  *)
    echo "Usage: $0 <command>"
    echo ""
    echo "Commands:"
    echo "  charge           Enable charging (will stop discharging if active)"
    echo "  +c               ^ Alias"
    echo "  stop-charge      Disable charging"
    echo "  -c               ^ Alias"
    echo "  discharge        Force battery discharging"
    echo "  +d               ^ Alias"
    echo "  discharge-off    Stop forced discharging"
    echo "  -d               ^ Alias"
    echo "  status           Show battery percentage, power source, and SMC charging state"
    echo "  s                ^ Alias"
    ;;
esac
```
I quickly got chatgpt to write up this code for me using the new keys. It works fine. I'll help integrate this into batt if needed

oliver408i, 2025-07-03:
Looks like these are the only two keys currently found, but they should be good enough to replace most functionality as needed. Unfortunately I do not know Go so I can't really help that much with integrating this into the actual code but I assume it should be pretty simple now that we know the keys and their values.

charlie0129, 2025-07-03:
I added the SMC keys for macOS Tahoe firmware in #52, thanks @oliver408i for the information. **I have no idea whether my implementation is correct or not**, so there won't be a release until it's tested.

If any volunteer wants to test it, you can build batt from source (Run `make build-gui` and binaries should be in `bin/`). Feedback is greatly appreciated!

peterneutron, 2025-07-07:
@charlie0129 Tested this today on an OG M1 MBP running Tahoe 26.0 Beta (25A5295e) with "Optimized Battery Charging" disabled. 

Confirmed working so far:

- Stop charging >= 80%
- Continue charging < 80%
- Keep charge at 80% with closed lid and connected to power supply

Not tested:

- Any limit other than 80%
- MagSafe LED control

charlie0129, 2025-07-10:
v0.4.0 has been released with experimental macOS Tahoe compatibility

https://github.com/charlie0129/batt/releases/tag/v0.4.0

peterneutron, 2025-07-13:
@charlie0129

On further testing I found an issue with the usage of `AdapterKeys` in that `CH0I` always reports `no data` and `CH0J` doesn't exist in Tahoe.

*   `CH0I` (`AdapterKey1`) consistently reports `no data`.
*   `CH0J` (`AdapterKey2`) missing entirely.

Because `AdapterKey1` is hardcoded in `adapter.go`, running `batt status` always fails with:

```
Error: failed to get power adapter status: failed to get power adapter status: got 500: "key has no data, check if it is valid"
```

I propose two potential solutions:

1.  Use `AC-W`
    This seems to be the most robust method. Using `ACPowerKey` (`AC-W`) would check if an adapter is present based on actual power draw. If `AC-W > 0`, the adapter is connected and in use.

2.  Use `CHCC`
    In my testing, `CHCC` provides a clean, binary status for the adapter connection. It accurately reports `1` when connected and `0` when disconnected, making it a reliable flag.
    *   Connected: `CHCC [ui8 ] 1 (bytes 01)`
    *   Disconnected: `CHCC [ui8 ] 0 (bytes 00)`

I couldn't quite discern if the goal was:

*   (A) To know if an adapter is connected at all?
*   (B) To know if charging is actively enabled or disabled while an adapter is connected?

charlie0129, 2025-07-13:
@peterneutron Thank you for your detailed report!

> I couldn't quite discern if the goal was

The goal of `CH0I` was to manually connect or disconnect the power adapter. Writing `0x1` to it will disconnect the power while writing `0x0` does the opposite.

Based on your report, it seems we have to find another key to disconnect/connect power adapters.

> (A) To know if an adapter is connected at all?

As you probably seen in the code, this is done using `AC-W`.

> `CH0J` doesn't exist in Tahoe

It seems [the comment above](https://github.com/charlie0129/batt/issues/34#issuecomment-2958460268) shows he has `CH0J`? Not sure what's going on.

If it exists, could you please check if `CH0J` works the same as `CH0I`? Like mentioned above (Writing `0x1` to it will disconnect the power while writing `0x0` does the opposite.)

peterneutron, 2025-07-13:
Sorry must have switched `CH0J` and `CH0I` in the above comment. Here is the actual output for all `CH0x` keys.

```
smc -l | grep CH0
  CH0D  [hex_]  (bytes 0c 00 00 00)
  CH0E  [hex_]  (bytes 00 00 00 00 00 00 00 00)
  CH0H  [ui32]  0 (bytes 00 00 00 00)
  CH0J  [    ]  no data
  CH0R  [hex_]  (bytes 00 00 00 00)
  CH0V  [ui32]  1378746368 (bytes 52 2e 00 00)
```

The `CH0J` key is neither directly readable nor writable

```
sudo smc -k CH0J -w "01"
Error: SMCWriteKey() = e00002c1
sudo smc -k CH0J -r
Error: SMCReadKey() = e00002c1

oliver408i, 2025-07-14:
> Sorry must have switched `CH0J` and `CH0I` in the above comment. Here is the actual output for all `CH0x` keys.
>
> ```
> smc -l | grep CH0
>   CH0D  [hex_]  (bytes 0c 00 00 00)
>   CH0E  [hex_]  (bytes 00 [...]

What version are you on? Beta 3?

My CH0J works fine.
  CH0J  [ui8 ]  0 (bytes 00)

Right now, do NOT update if you want basically any battery control software to work. Until Apples get a stable Tahoe release I highly advise you not to update if you depend on batt and similar software.

oliver408i, 2025-07-14:
CHCC indicated adapter state only (it follows CH0J) but it cannot be edited to control discharging, unlike CH0J.

oliver408i, 2025-07-14:
Hello, seems like we have (yet another) new key for discharge control: `CHIE`
`sudo smc -k CHIE -w 08` activates discharge
`sudo smc -k CHIE -w 00` disables it

oliver408i, 2025-07-14:
God they are really playing with us. Seems like `CHIE` is working on beta 2 and beta 3. Charge control seems to remain same on `CHTE`.

peterneutron, 2025-07-14:
> What version are you on? Beta 3?

Yes I'am running Beta 3. CH0J is no dice here.

>  Hello, seems like we have (yet another) new key for discharge control: CHIE

But this works perfectly here with a patched version of batt using `CHTE` and `CHIE`. Thanks a lot.

charlie0129, 2025-07-14:
@peterneutron Hi, v0.4.1 is released with `Force Dischage`

https://github.com/charlie0129/batt/releases/tag/v0.4.1

charlie0129, 2025-08-01:
Closing because there are no known issues with macOS 26 so far.

closed 2025-08-01 by charlie0129

mhaeuser, 2025-08-06:
@charlie0129 I guess the SMC firmware is now stable, 15.7 RC rolls this out.