charlie0129/batt #35: MacBook Pro often does not charge when plugged in

nhojb, 2025-06-11:
I have `batt` setup with an 80% charging limit. Even though the status says that charging is enabled it very often does not charge at all when plugged in and sleeping. The limit is great, but it's rather annoying to come back to my MBP and find it hasn't charged at all 🫤

macOS: 15.5
batt: v0.3.7

[image]

charlie0129, 2025-06-11:
Charging is disabled during sleep by default. If you want to charge during sleep, this behavior can be changed at the risk of overcharging.

See https://github.com/charlie0129/batt?tab=readme-ov-file#why-does-my-macbook-stop-charging-after-i-close-the-lid

nhojb, 2025-06-11:
Makes sense. Thank you for the quick response!

closed 2025-06-11 by nhojb