charlie0129/batt #37: Allow batt with lid closed on ac

MabaKalox, 2025-06-20:
Note - idea discussion, not feature request.

I typically charge MacBook with lid closed (cat runs over it, etc.). As I understood currently batt does not support such use case, since MacBook will go to sleep.
But I step onto commands:
`pmset -c disablesleep 1`, which prevents sleep *while on charger* (even with lid closed).
`pmset -a acwake 1`, which supposed to wake on bat/ac change.
Could it be used, instead of disable-charging-pre-sleep?
So flow could be:
disable-charging-pre-sleep handler would do instead:
1) pmset -c disablesleep 1
2) pmset -a acwake 1
3) *enable charger* (ac could be connected/disconnected currently)

sleeping Macbook -> attaching charger -> wake up due to AC wake -> check charge level -> 
a) above threshold -> disable charger (supposedly will go to sleep, since `disablesleep` with `-c` only works while charger connected)
b) below threshold -> enable charger -> charge threshold reached -> do (a)

On wakeup -> `pmset -a acwake 0` (not sure if needed)

I could try to implement POC, but first would like to hear if it could work in theory, or such idea already was considered but too complicated/not reliable/did no work?

MabaKalox, 2025-06-20:
hm, first issue - `pmset -c disablesleep 1` disables sleep both on battery and ac... (man page is missleading).

MabaKalox, 2025-06-20:
tried `caffeinate -s` (prevents sleep, works only on AC) with `pmset -a acwake 1`.
1) run `pmset -a acwake 1` and `caffeinate -s`
2) close lid -> sleep activates
3) connect charger -> acwake triggers -> caffeinate seems to prevent sleep.
Also screen saver continues to function as expected.

charlie0129, 2025-06-23:
> caffeinate -s seems to consistently block sleep while AC connected

Great, seems I am misled by the sleep notifications sent by macOS. macOS will still send sleep notifications while the mac itself does not sleep

MabaKalox, 2025-06-26:
Ups, accidentally closed, reopen please.

charlie0129, 2025-06-27:
Sure, will keep this issue open

PR #51 (2025-08-03): Add prevent-system-sleep while charging
This Pull request:
1) adds config option preventSystemSleep, which when enabled, will create Power Management Assertion of type [kIOPMAssertionTypePreventSystemSleep](https://developer.apple.com/documentation/iokit/kiopmassertiontypepreventsystemsleep) when batt enables charging. This assertion prevents system sleep while on AC (on battery it has no effect), forcing indefinite Dark-Wake state.
This allows batt to continue manage charging even in forced sleep mode (e.g. lid closed).

2) moves maintainLoop lock into inner loop, so it is prevents concurrent [... cut]
closed 2025-08-03 by charlie0129