charlie0129/batt #49: About the Type-C external monitor issue.

Eivs, 2025-06-24:
Hi, I really like this clean battery management tool. However, when using my MacBook in clamshell mode with a single USB-C cable connected to an external monitor (for both display and power), I encounter the following issues:  

After enabling `batt limit 80`, although the current battery is already at 100%, it still shows as charging. Even after 2 hours of continuous use, the battery doesn’t discharge and remains at 100%.  

I tried disabling the adapter with `batt adapter disable`, which stops the charging, but then my external monitor stops working. The display only works normally after re-enabling it with `batt adapter enable`.

charlie0129, 2025-06-24:
See #11 #26 #37

> Even after 2 hours of continuous use, the battery doesn’t discharge and remains at 100%.

Yes, it won't discharge. Unless `batt adapter disable`.

> but then my external monitor stops working

This is Apple's limitation. Overriding this limitation requires some hacks, and batt haven't implemented anything to overcome this limitation to make sure your MacBook behaves like normal (there are some on-going progresses by community members, see the issues).

charlie0129, 2025-06-24:
Great. If it can work, you can report back in this issue so other users can know!