charlie0129/batt #53: cannot get batt status

ieuforu, 2025-07-14:
你好，安装的是cli。使用 batt limit 78 命令，显示成功。但是 batt status ：$ batt status
Error: failed to get power adapter status: failed to get power adapter status: got 500: "key has no data, check if it is valid"
。昨晚睡觉前搞得，早上起来确实暂停充电了。

charlie0129, 2025-07-14:
> Error: failed to get power adapter status: failed to get power adapter status: got 500: "key has no data, check if it is valid"

Are you using macOS Tahoe? This is an known issue.

> 是不是M芯片只有80 和 100 这连个预设值

Charge limit can be anything between 20 - 100.

ieuforu, 2025-07-14:
No, I'm using macOS 15 and it's paused until 80%

ieuforu, 2025-07-14:
How can I make him pause at 70. The command has been executed,
$ batt limit 70
INFO[2025-07-14T15:03:45+08:00] daemon responded: "set upper/lower charging limit to 70%/68%, current charge: 80%. Current charge is above the limit, so your computer will use power from the wall only. Battery charge will remain the same."
INFO[2025-07-14T15:03:45+08:00] successfully set battery charge limit to 70%

How long will it take to see the effect?

charlie0129, 2025-07-14:
> Updated Tahoe, but downgraded to sequoia after wiping the disk.

You are using Tahoe firmware. No matter how you wiped the disk. There is no going back without DFU.

> Currently, it can only pause at 80%, despite using the bat limit 70 command

Unplug the power adapter.

charlie0129, 2025-07-14:
Please read the README.

closed 2025-07-14 by charlie0129

charlie0129, 2025-07-15:
Please use v0.4.1, discharge support in Taboe is included. You are using Tahoe firmware, not Sequoia.

ieuforu, 2025-07-15:
目前有时候显示 ”电池没有在充电“

ieuforu, 2025-07-15:
OK， I solved it. Also, I tried installing with brew and it didn't report any errors. The CLI version is ready. I used the bat limit 70 command and checked the effect at night

ieuforu, 2025-07-15:
Now it still shows' battery not charging ', which the official said was a problem before. Is this normal after using bat? Shouldn't it display 'pause charging' like the official

charlie0129, 2025-07-15:
It's expected.

ieuforu, 2025-07-15:
okey，OK， I have thoroughly resolved all doubts. Thank you for this exquisite tool, which allows me to control the battery anytime and anywhere, instead of having to spend time studying again after unplugging the power. You can delete this issue now!