charlie0129/batt #57: Unable to open 0.4.1 on MacBook air

weinhsu-usc, 2025-08-04:
Hey there, first of all cool product! This is indeed one of the best alternatives for mac battery management, especially as there is an Raycast extension for it as well. However, I could not open the app after downloading the .dmg file. Here is my mac:

[image]

I followed the instructions and entered sudo xattr -r -d com.apple.quarantine /Applications/batt.app into the terminal, but when I double clicked on the app nothing happened. The app also has no icon, but not sure if that's by design.

[image]

The app clearly opened because if I try to move it to trash, it won't let me "because it's open", but besides existing simply existing in my activity monitor I can see no other signs of it being open nor interact with it. I would love to come back to use batt if this is resolved.

charlie0129, 2025-08-05:
> especially as there is an Raycast extension for it as well

thank you for your information. apparently that's news to me, i am not aware of the raycast extension

charlie0129, 2025-08-08:
> Could you provide a picture of the menubar icon

You can refer to the screenshot in README, it shows the icon very clearly, in great detail.

> yielded command not found

that's because it not installed yet (install buttom is in the menubar app)

weinhsu-usc, 2025-08-09:
Got it, turns out I'm using a menu bar manager app, which had some bug that kept new applications hidden.  I saw the batt icon momentarily when I restarted my mac once which led me to investigate. Sorry for taking up your time with an issue that wasn't your fault!

closed 2025-08-09 by weinhsu-usc