charlie0129/batt #61: Error running v0.5.0 on macOS Tahoe

Xerxes-2, 2025-08-22:
`batt status` shows

```
Error: failed to get battery info: failed to unmarshal battery info: json: cannot unmarshal number 74910.825 into Go struct field Battery.Design of type int
```

Machine: MacBook Pro 14" M2 Pro 10-Core
OS: macOS 26.0 Beta 7
batt version: v0.5.0

charlie0129, 2025-08-22:
@peterneutron could you help diagnose what the issue is?

charlie0129, 2025-08-22:
@Xerxes-2 Thank you for your report! Meanwhile, maybe you can revert to the old version

charlie0129, 2025-08-22:
@peterneutron Never mind, I may know the cause.

@Xerxes-2 Did you upgrade the daemon? It looks like you did not upgrade the daemon. You can refer the release page for how to upgrade.

Xerxes-2, 2025-08-22:
> [@peterneutron](https://github.com/peterneutron) Never mind, I may know the cause.
> 
> [@Xerxes-2](https://github.com/Xerxes-2) Did you upgrade the daemon? It looks like you did not upgrade the daemon. You can refer the release page for how to upgrade.

Problem solved.
Could you update the homebrew caveats? I see currently there is lacking a bootstrap guide, especially the service must run with `sudo`

closed 2025-08-22 by Xerxes-2

charlie0129, 2025-08-22:
Well, I tried, the Homebrew guys prefer not to add such caveats. Also, in earlier Homebrew versions, `must run with sudo` is automatically displayed by Homebrew itself. I am not sure what's wrong with Homebrew, it no longer shows `must run with sudo`.

peterneutron, 2025-08-22:
@charlie0129 Seems to be bug or intentional behaviour on part of Homebrew, when installing directly from a local .rb (which is no longer possible without creating a local tap instead of just pointing it to a file) the `sudo` is displayed on install.

But when using the JSON API the `require_root` is skipped when `run` is not present in the `service` section. Relevant code is at `Homebrew/service.rb` line 513 `return { name: name_params }.compact_blank if @run_params.blank?`.

Sometimes I imagine what the future would've looked like if we all had embraced a pacman port instead of homebrew 😅