charlie0129/batt #62: An option to hide the menu bar icon (Batt GUI)

qnblackcat, 2025-08-25:
First, I have to say thank you for creating this app. It's simple yet has everything I want. 

However, it could be better if we had an option to hide the menu bar icon. I always set the charge limit to 80% and have literally no reason to open Batt for anything else. This could make my menu bar appear cleaner.

Thanks in advance!

charlie0129, 2025-08-25:
Yes, you can. You have two options:

1. use the command line version. there is no gui, so no menubar icon.
2. just quit the menubar app. it will continue to function

charlie0129, 2025-08-25:
The menubar app is just a client. The actual charging control lives in the batt daemon, not the app, so the app can be quit safely.