charlie0129/batt #64: Something is wrong with the display "Full capacity"...

ulrichloose, 2025-08-31:
ioreg -w 0 -f -r -c AppleSmartBattery
...
"DesignCapacity" = 4563
"AppleRawMaxCapacity" = 4542
...

Battery status:
  Current charge: 80%
  State: not charging
  Full capacity: 4.6 Wh
  Charge rate: 0.0 W
  Voltage: 12.29 V

On the one hand, the value "DesignCapacity" is apparently used and not the currently valid value "AppleRawMaxCapacity" which takes the aging into account. On the other hand, these values are data in mAh and not mWh... It would be better not to evaluate "DesignCapacity" but "AppleRawMaxCapacity" and you would have to display the displayed value in Ah. Alternatively, you would have to multiply with approx. 11.5 to get to the Wh...

Basically it doesn't matter - but if you have something else to do, you could fix this tiny discrepancy.

peterneutron, 2025-09-01:
@ulrichloose did you update your daemon?

> Full capacity: 4.6 Wh
> Charge rate: 24.1 W

This cannot be correct I changed the unit of `Full capacity`  to `mAh` and the `Charge rate` should have a sign `+/-`

[image]

charlie0129, 2025-09-02:
@ulrichloose Can you reliably reproduce it? or occasionally?

ulrichloose, 2025-09-02:
First  - with the changes by peterneutron everything works on my machine as expected. I cloned the repository newley, changed the the files und run make app...

change to /bin and run ./batt status works, and ./batt gui works and double-click the App also works.

Before these changes the app did not start and I tried ./batt gui for some errormessages...

ulrichloose, 2025-09-02:
Now something completely different...

I didn't leave data.batteryInfo.Design alone for display and calculation in status.go. I have no idea about Go, but the health percentage is displayed in the GUI and could be used to calculate the actual battery capacity. I just cheekily changed / added a few lines...

// Battery Info.
cmd.Println(bold("Battery status:"))

snapshot, err := apiClient.GetPowerTelemetry()
if err != nil {
// The error handling is missing here...
}

maxCapacitymAh := float64(data.batteryInfo.Design) / 100.0 * float64(snapshot.Calculations.HealthByMaxCapacity)
..
..
if data.batteryInfo.State == powerinfo.Charging && config.UpperLimit() < 100 && data.currentCharge < config.UpperLimit() {
// Work in mAh directly (no Wh conversions)
targetCapacitymAh := float64(config.UpperLimit()) / 100.0 * maxCapacitymAh
currentCapacitymAh := float64(data.currentCharge) / 100.0 * maxCapacitymAh
capacityToChargemAh := targetCapacitymAh - currentCapacitymAh
..
..
cmd.Printf("  State: %s\n", bold("%s", state))
cmd.Printf("  Max Capacity: %s\n", bold("%.0f mAh", maxCapacitymAh)) 

Of course, this is all absolutely lay, but it works. With a new computer with a 100% battery, none of this matters, but after a few years it goes towards 80% and then the calculations and displays become more and more inaccurate. Maybe a professional can implement this correctly?

ulrichloose, 2025-09-02:
peterneutron

everything works - but... the menu and submenu is not very readable...

[image]