charlie0129/batt #68: Daemon runs with debug logging by default

opirko, 2025-09-16:
Just installed `batt` via install.sh script. I've noticed that the started daemon is logging `debug` by default. I suspect this is unintended. I'd expect `info` to be the default, so maybe running the Daemon without log level specified, then it should default to `info`?
```
0  3778     1   0  7:23PM ??         0:00.06 /usr/local/bin/batt daemon --log-level=debug
```

charlie0129, 2025-09-17:
It's expected actually.

1. debug log volume is not that larger than info level. Many macOS system components have a much larger log volume so I don't expect it to cause any issue.
2. it makes debugging so much easier. users can report problems without turning on debug logs first, which will miss the opportunity of reporting it as some problems are hard to reproduce.

opirko, 2025-09-17:
I see your point. On the other hand if every SW component by default printed debug lvl logs, it may add up and will unnecessarily wear out the SSD. Either way it's a judgement call. Thanks for the feedback.