charlie0129/batt #71: Error: failed to get charging status: failed to get charging status: got 500: "key has no data, check if it is valid"

AizazZaidee, 2025-09-18:
sudo brew services start batt
Service `batt` already started, use `brew services restart batt` to restart.

sudo batt limit 60
INFO[2025-09-18T15:26:30+02:00] daemon responded: "set upper/lower charging limit to 60%/58%, current charge: 100%. Current charge is above the limit, so your computer will use power from the wall only. Battery charge will remain the same." 
INFO[2025-09-18T15:26:30+02:00] successfully set battery charge limit to 60% 

sudo batt prevent-idle-sleep enable
INFO[2025-09-18T15:26:47+02:00] daemon responded: "ok"                       
INFO[2025-09-18T15:26:47+02:00] successfully enabled prevent-idle-sleep  

sudo batt disable-charging-pre-sleep enable
INFO[2025-09-18T15:26:58+02:00] daemon responded: "ok"                       
INFO[2025-09-18T15:26:58+02:00] successfully enabled disable-charging-pre-sleep 

### sudo batt status
Error: failed to get charging status: failed to get charging status: got 500: "key has no data, check if it is valid"

It stops working, now my battery is charging 100%

### Info
Darwin fki-MacBook-Pro.local 24.6.0 Darwin Kernel Version 24.6.0: Mon Aug 11 21:16:34 PDT 2025; root:xnu-11417.140.69.701.11~1/RELEASE_ARM64_T6020 arm64

system_profiler SPHardwareDataType | grep -i 'Firmware Version'
      System Firmware Version: 13822.1.2

`time="2025-09-18 15:09:17.727" level=info msg="batt version v0.2.1 commit 90df4aaefb930ab05a6763c96866e8db0faf698e"
time="2025-09-18 15:09:17.727" level=info msg="config loaded: main.Config{Limit:60, PreventIdleSleep:true, DisableChargingPreSleep:false, AllowNonRootAccess:false, LowerLimitDelta:2, ControlMagSafeLED:false}"
time="2025-09-18 15:09:17.728" level=info msg="http server listening on /var/run/batt.sock"
time="2025-09-18 15:09:17.728" level=info msg="registered and listening system sleep notifications"
time="2025-09-18 15:09:17.728" level=debug msg="main loop starts"
time="2025-09-18 15:09:17.729" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:09:37.729" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:09:57.730" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:10:17.732" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:10:37.733" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:10:57.734" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:11:09.017" level=error msg="getCharging failed: key has no data, check if it is valid"
time="2025-09-18 15:11:09.017" level=error msg="Error #01: key has no data, check if it is valid\n" dataLength=39 latency=1 method=GET path=/charging statusCode=500
time="2025-09-18 15:11:17.735" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:11:37.735" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:11:57.738" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:12:17.739" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:12:37.740" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:12:38.294" level=info msg="set charging limit to 60"
time="2025-09-18 15:12:38.294" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:12:38.295" level=debug msg="PUT /limit 201 (2ms)" dataLength=190 latency=2 method=PUT path=/limit statusCode=201
time="2025-09-18 15:12:56.024" level=error msg="getCharging failed: key has no data, check if it is valid"
time="2025-09-18 15:12:56.024" level=error msg="Error #01: key has no data, check if it is valid\n" dataLength=39 latency=1 method=GET path=/charging statusCode=500
time="2025-09-18 15:12:57.741" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:13:17.743" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:13:37.744" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:13:57.746" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:14:17.747" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:14:37.748" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:14:57.749" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:15:17.751" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:15:37.752" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:15:57.754" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:16:17.754" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:16:37.755" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:16:57.756" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:17:17.757" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:17:37.760" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:17:57.761" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:18:17.761" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:18:37.763" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:18:57.764" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:19:17.765" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:19:37.767" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
[... 30 lines cut ...]
time="2025-09-18 15:27:18.679" level=error msg="getCharging failed: key has no data, check if it is valid"
time="2025-09-18 15:27:18.679" level=error msg="Error #01: key has no data, check if it is valid\n" dataLength=39 latency=1 method=GET path=/charging statusCode=500
time="2025-09-18 15:27:37.816" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:27:57.817" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:28:17.818" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:28:37.820" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:28:57.821" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:29:17.822" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:29:37.824" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:29:57.825" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:30:17.827" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:30:37.827" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:30:57.829" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:31:17.830" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:31:37.832" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:31:57.833" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:32:17.835" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:32:37.836" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
[... cut ...]
time="2025-09-18 15:33:17.838" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:33:37.840" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:33:57.841" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:34:17.843" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
time="2025-09-18 15:34:37.844" level=error msg="IsChargingEnabled failed: key has no data, check if it is valid"
`

charlie0129, 2025-09-19:
> batt version (daemon log): v0.2.1

This is too old (it will not work on 13822).

Try the latest version instead.

AizazZaidee, 2025-09-19:
Thanks @charlie0129 I have updated to v0.5.2 c68212d343e99de2c383ab934ddc36ee6046dded and I still have the same issue 
sudo batt status
Error: failed to get charging status: failed to get charging status: got 500: "key has no data, check if it is valid"

charlie0129, 2025-09-19:
Check /tmp/batt.logs for the version, to see if it's actually updated