charlie0129/batt #74: question: battery is actively discharging when plugged in and between upper/lower limits

g-a-c, 2025-09-25:
```
Charging status:
  Allow charging: ✘ (refreshes can take up to 2 minutes)
    Your Mac will not charge even if you plug in, because your current charge is above the lower limit. Charging will be allowed after current charge drops below the lower limit.
  Use power adapter: ✔
    Your Mac will use power from the wall (to operate or charge), if it is plugged in.

Battery status:
  Current charge: 70%
  State: discharging
  Full capacity: 5815 mAh
  Charge rate: -0.7 W
  Voltage: 11.91 V

Battery configuration:
  Upper limit: 75%
  Lower limit: 70%
  Prevent idle-sleep when charging: ✘
  Disable charging before sleep if charge limit is enabled: ✘
  Prevent system-sleep when charging: ✔
  Allow non-root users to access the daemon: ✔
  Control MagSafe LED: ✔
```
Macbook Pro M2 14-inch 2023, macOS 26 Tahoe

I use 75/70 as my upper and lower limit because sometimes I do use my laptop for a short period disconnected from mains power and it doesn't drop a lot, so I don't see the point always keeping it topped up to 75%.

I noticed today that even though my laptop is currently plugged in with a Ugreen 100W USB-C PD power brick and an Apple USB-C cable, the battery status is showing as actively `discharging` with a negative charge rate. I'm sure it should show `not charging` with a `0` charge rate?

I don't know if this started with macOS 26, I only noticed it for the first time today, but I don't know whether this was also happening before the upgrade. It's as if `batt` has some kind of battery "floating" where it will allow discharging to the lower limit, and then charge to the upper limit - but I don't think it has such functionality?

g-a-c, 2025-09-25:
> Will it keep discharging until below 70%?

Yes - at some point it dipped below 70%. Now it shows that it's slowly charging back up again:

```
Battery status:
  Current charge: 72%
  Time to limit (75%): ~69 minutes
  State: charging
  Full capacity: 5820 mAh
  Charge rate: +1.8 W
  Voltage: 12.06 V
```

> Are you plugged in all the time when discover it

I haven't used the laptop on battery for a few days. It was plugged in when I noticed this issue this morning, I had moved rooms at some point but I unplugged it in one room and plugged it in in the other.

g-a-c, 2025-09-25:
> Can you check if Apple’s Optimized Charging is enabled or not

I turned this off months ago when I started using `batt` - I just double checked to make sure the Tahoe upgrade didn't re-enable it but no; it is still disabled.

charlie0129, 2025-09-25:
I am not sure about the problem currently. I am not using macOS 26 so not sure if it’s related to it either.

Let’s see if it happens often or if it happened to somebody else too.

g-a-c, 2025-09-25:
It's now finished charging to 75% and is now discharging again:

```
Battery status:
  Current charge: 75%
  State: discharging
  Full capacity: 5821 mAh
  Charge rate: -1.1 W
  Voltage: 12.11 V
```

I can only assume that it's either related to macOS 26, or it's related to the fact that I'm charging over USB-C and not Magsafe. Will wait to see if others have issues.

g-a-c, 2025-09-25:
I've had the `batt status` refreshing every 30 seconds to watch the behaviour, so I tried to remove all the `GET` requests which were related to that, and this is what I'm left with for the time period since I opened this ticket (during which the battery discharged below 70, then charged back up to 75, then started discharging again)

```
time="2025-09-25T14:44:04+01:00" level=debug msg="maintain loop status" batteryCharge=69 chargingEnabled=false isPluggedIn=true lower=70 maintainedChargingInProgress=false upper=75
time="2025-09-25T14:44:04+01:00" level=info msg="Battery charge is below lower limit, enabling charging" batteryCharge=69 lower=70 upper=75
time="2025-09-25T14:44:14+01:00" level=debug msg="maintain loop status" batteryCharge=69 chargingEnabled=true isPluggedIn=true lower=70 maintainedChargingInProgress=true upper=75
time="2025-09-25T15:06:55+01:00" level=debug msg="maintain loop status" batteryCharge=70 chargingEnabled=true isPluggedIn=true lower=70 maintainedChargingInProgress=true upper=75
time="2025-09-25T15:13:45+01:00" level=debug msg="maintain loop status" batteryCharge=71 chargingEnabled=true isPluggedIn=true lower=70 maintainedChargingInProgress=true upper=75
time="2025-09-25T15:21:05+01:00" level=debug msg="maintain loop status" batteryCharge=72 chargingEnabled=true isPluggedIn=true lower=70 maintainedChargingInProgress=true upper=75
time="2025-09-25T15:29:45+01:00" level=debug msg="maintain loop status" batteryCharge=73 chargingEnabled=true isPluggedIn=true lower=70 maintainedChargingInProgress=true upper=75
time="2025-09-25T15:34:55+01:00" level=debug msg="maintain loop status" batteryCharge=74 chargingEnabled=true isPluggedIn=true lower=70 maintainedChargingInProgress=true upper=75
time="2025-09-25T15:38:35+01:00" level=debug msg="maintain loop status" batteryCharge=75 chargingEnabled=true isPluggedIn=true lower=70 maintainedChargingInProgress=true upper=75
time="2025-09-25T15:38:35+01:00" level=info msg="Battery charge is above upper limit, disabling charging" batteryCharge=75 lower=70 upper=75
time="2025-09-25T15:38:45+01:00" level=debug msg="maintain loop status" batteryCharge=75 chargingEnabled=false isPluggedIn=true lower=70 maintainedChargingInProgress=false upper=75
```

g-a-c, 2025-09-25:
Interesting. The power brick is a Ugreen Nexode GaN 100W charger, with only this laptop plugged into it.

However, I just checked the System Information application and it shows that there's only 22W of power being detected:

```
AC Charger Information:

  Connected:	Yes
  ID:	0x0000
  Wattage (W):	22
  Family:	0xe000400a
  Charging:	No
```

With this charger, and the stuff I have plugged into it, the port I'm using should be able to use up to the full 100W since nothing else is plugged in. I'm 99% sure this is an Apple USB-C cable - but I wonder if it's actually from my iPhone and can't deliver the full charge to my Macbook. I'll try switching out the cable and see what happens, and that may explain the slow charging and _potentially_ the power drain (if my laptop is consuming more than 22W while running, I don't know what its idle power consumption is).

I'll try and find an alternative cable and see if I can get System Information to report a higher wattage (I think the charger that comes with a 14" model is 65W so I should at least be able to see that even if not the full 100W) and see what happens.

g-a-c, 2025-09-25:
Argh. OK, I'm an idiot.

There was only one device connected to the charger, but there were two cables; the USB-C one I'm using, and the USB-C/Lightning one that charges my AirPods (not currently plugged in).

I had them mixed up at the charger end - the Lightning cable was plugged into the high wattage port, the USB-C was plugged into the low power one. Quick switch and now System Information shows that there's 65W of power available to my laptop which is more like what I expected to see.

My bad! I'll close this issue but definitely worth leaving up, maybe someone else will learn from my mistakes 🙄😂

closed 2025-09-25 by g-a-c